PX4/PX4-Autopilot #11346: Teraranger driver incompatibility with pixhawk 4

vivekalgorythma, 2019-01-30:
If you connect the teraranger one to the i2c port nothing happens. 

This is due to the driver only looking at PX4_I2C_BUS_EXPANSION

This port is not the i2c portA on the pixhawk 4 and hence does not work. 
To fix this the driver needs to look at PX4_I2C_BUS_EXPANSION2 instead. 

I have made a commit here 
https://github.com/algorythma/Firmware/commit/2d085150c4201fd1eaae40c4fee4dfde98f8670f

Could someone please fix this in the main repo

dagar, 2019-01-30:
@vivekalgorythma the proper fix would be to try all external i2c buses. If you're interested in doing this in a PR feel free to ping me if you have any questions.

vivekalgorythma, 2019-01-31:
@dagar I just cloned the master branch and it seems to do as you say in that. I was using the stable 1.8.2 version. I dont think this is necessary anymore

closed 2019-07-08 by stale[bot]