PX4/PX4-Autopilot #11355: GPS and vision data for pose estimation

NicoNelli, 2019-01-31:
Hi everyone,
I am using a drone equipped with PixHawk and an onboard computer. I was using LPE and now I have  switched to EKF2 as @TSC21 told me. I'd like to fuse GPS and vision data coming from an onboard vision system. My purpose is to use GPS data and, during complex maneuvers, I want to add vision data to improve the state estimation.
May I change in flight the parameter EK2_AID_MASK to implement such behaviour?
I saw that it requires a reboot vehicle.
Any suggestions?

Regarding the addition of the vision data, I find out two mavros topics:

/mavros/vision_pose/pose

/mavros/odometry/odom

I guess the first one regards absolute measurement and the second one regards vehicle's perspective, is not true ?

Thank you for your help

TSC21, 2019-01-31:
>May I change in flight the parameter EK2_AID_MASK to implement such behaviour?

No you can't do that. You will need to set the parameter to fuse both vision and GPS and stop the vision stream when you don't want it to be used. I don't see why though you wouldn't be using both always.

>/mavros/vision_pose/pose
>
>/mavros/odometry/odom

You are not using the latest MAVROS release, since the latest is now odometry/out. Also it's pretty clear what they represent: one is pose with cross-covariance for pose, while the other is a full odometry message containing also linear and angular velocities with respective covariances (if your estimator is capable of throwing that as well).

TSC21, 2019-01-31:
Also both topics should represent egomotion estimation.

NicoNelli, 2019-01-31:
Thank you @TSC21. I have a landing platform with some tags on top and I am using them to improve the accuracy of the state estimation during the landing procedure. During other flight actions, I do not have such information. Your solution to stop vision stream should be fine for my case

NicoNelli, 2019-02-05:
Hi everyone, 
As suggested by @TSC21, I switched to EFK2 and I set these parameters:

EFK2_AID_MASK = 9 (vision and GPS data)
EKF2_EVP_NOISE = 0.20 m
EKF2_EV_GATE = 5 SD
EKF2_GPS_P_GATE = 5 SD
EKF2_GPS_V_NOISE = 3 m
EKF2_HGT_MODE = vision

However with these parameters, the EFK2 follows x and y GPS data and z vision data. It should always follow vision data if they are available. 

 But if I use only vision data( EFK2_AID_MASK = 8 ), the EKF2 works well.

Any suggestion?

Thank you very much for your time

TobiMiller, 2019-02-07:
how is the vision feed handled inside of px4 stack? as local or global position? what will happen when i send the MAVs position relative from the marker as vision?

NicoNelli, 2019-02-11:
hi @TobiMiller. What I am doing is a visual inertial odometry and I inject it into the px4 via mavros package. The vision data, basically, are taken into account in the state estimation. With relative position from markers, you can do the same.

NicoNelli, 2019-02-19:
I have always the same problems, any advice ?
Thank you

TobiMiller, 2019-02-19:
@NicoNelli @TSC21 
I don't have a problem. I just want to understand the EKF exactly. The PX4 firmware sets the first GPS coordinate as reference for the local position. if i only set the vision in the mask, then there is no reference to the world coordinate system. 

my question is now. what happens if i merge GPS and vision and come from outside into a room without GPS and feed vision odometry? where is my reference then? if you set target values, then they are always in the local frame. 

Example:
- i only start with GPS information outside
- i give setpoint positions from my local frame located at my start position
- I'm flying into a building and starting visual odometry and the gps signal is breaking off
- does the local frame jump to my odometry frame now? because it's important to know when specifying your setpoint position

regards

rohantiw96, 2019-03-15:
I have the same doubt as @TobiMiller.
@TSC21 Any views on this ?

closed 2019-07-08 by stale[bot]