PX4/PX4-Autopilot #11359: EKF2 local_position doesn't initialize without mag sensor

ArkadiuszNiemiec, 2019-01-31:
**Describe the bug**
I am trying to use EKF2 indoors with lidar, external vision system and optical flow. I had to disable all mag sensors because of strong magnetic interference and since then the PX4 is not publishing `local_position` messages. I have an angle published on `mavros/vision_pose/pose` and `vision_yaw_fusion` enabled in `ekf2_aid_mask`. Firmware version is 1.8.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable all mag sensors
2. Enable `vision_yaw_fusion` in `ekf2_aid_mask` and publish real data on `mavros/vision_pose/pose` topic.
3. Reboot and watch that no `local_position` messages are published.
4. See error

**Expected behavior**
I expect the EKF2 to initilize without a mag sensor and with angle from external vision system.

ssingh19, 2019-02-05:
Can confirm this behavior exists with Firmware 1.7.3. I have vicon_bridge running plus an additional node converting the vicon tf info to /mavros/vision_pose/pose. 

On Pixhawk, I have EKF2_MAG_TYPE set to none, and the ekf2_aid_mask set as above. 

No /local_position/pose messages coming back from pixhawk :(

Arkadiusz - what happens if you enable a mag sensor and still keep the vision_yaw_fusion bit in ekf2_aid_mask set to true?

ArkadiuszNiemiec, 2019-02-05:
@ssingh19 it works. I have decreased the `EKF2_MAG_GATE` and increased `EKF2_MAG_NOISE` and `EKF2_MAG_E_NOISE` so that Kalman will less rely on mag sensor.

ssingh19, 2019-02-05:
@ArkadiuszNiemiec  Thanks for letting me know. It would be nice if EKF2 knows how to handle EKF2_MAG_TYPE set to none and vision_yaw_fusion enabled.

ssingh19, 2019-02-05:
Another observation. If I don't connect a GPS, then /mavros/local_position/pose is again not published. Using qgc analyze, LOCAL_POSITION_NED (the routed message) is not even generated. So it appears, that *at least* a GPS is necessary, even if Vicon info is being provided. 

Currently, a GPS with external mag is connected and all messages are being published. Although the local_position/pose is well off - showing values like (x.xxx e-17 and x.xxx e-33). Should be due to tuning of the EKF2 params.

ArkadiuszNiemiec, 2019-02-06:
> Another observation. If I don't connect a GPS, then /mavros/local_position/pose is again not published. Using qgc analyze, LOCAL_POSITION_NED (the routed message) is not even generated. So it [...]

I am pretty sure that my setup works without a GPS connected. After publishing few messages on `mavros/vision_pose/pose` (with mag0 enabled) the local position is appearing.

ArkadiuszNiemiec, 2019-02-17:
Ok, so I was able to fly without a compass and without a vision enabled but not being published. This is

ArkadiuszNiemiec, 2019-02-18:
I have tracked the issue to the method `Ekf::initialiseFilter()` of `Ekf` in `ecl` library. During the initialization it checks if `mag` data is coming. I have edited this line to omit this step and was able to fly without a mag sensor and without external vision data stream.

In the stable version it also checks the external vision and this was preventing the EKF to initialize without incoming vision data stream when fusion is enbaled:
```
	if (hgt_count_fail || mag_count_fail || ev_count_fail) {
```
In current master it's reduced to only check `hgt` and `mag`:
https://github.com/PX4/ecl/blob/1378ec179717448404c74819871d149217350b19/EKF/ekf.cpp#L218

I think that it would be better to check the `EKF2_MAG_TYPE` and not check `mag_count_fail` if set to `5 (None)`. I will prepare a pull request.

ArkadiuszNiemiec, 2019-02-18:
I have tracked the issue to the method `Ekf::initialiseFilter()` of `Ekf` in `ecl` library. Master build of the PX4 fixes the problem with External Vision data being required to initialize the EKF2 when fusion is enabled but does not allow to run the estimator without any magnetometer on board.

During the initialization the ekf checks if `mag` data is coming:
https://github.com/PX4/ecl/blob/1378ec179717448404c74819871d149217350b19/EKF/ekf.cpp#L218-L221
I have edited this line to omit this step and was able to fly without any problems.
I think that it would be better to check the `EKF2_MAG_TYPE` and not check `mag_count_fail` if set to `5 (None)`. I will prepare a pull request.

JoeyLeeNPU, 2019-03-27:
Hello, I hear that 1.9 firmware fix this bug, have you try it? 
In firmware 1.8, I also meet a problem which is similar to yours. When I use slam_pose instead of GPS_based, in mavros/local_position/pose topic, there is no yaw infomation published to the topic. Because I want to use the local yaw (ENU) as a reference angle to transform other frames. 
I think the reason is that EKF doesn't work without some information.
In my project, I use both compasses in GPS and px4.

ArkadiuszNiemiec, 2019-03-27:
Yes, I am using 1.9 now and I was able to initialize it without a mag sensor. I am now heavily modifying the firmware to fly indoors and will PR if something useful comes out of it. I will close the issue for now. Thanks.

closed 2019-03-27 by ArkadiuszNiemiec