PX4/PX4-Autopilot #11361: "tests time" hangs and fails to recover in current Firmware/master

mcsauder, 2019-02-01:
**Describe the bug**
From a console, run "tests time" the test will hang indefinitely.

**To Reproduce**
From a console, run "tests time" the test will hang indefinitely.

**Expected behavior**
The test should execute properly. :)

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.
[image]

**Drone (please complete the following information):**
pixhawk4 (fmu-v5)

Please let me know if I can provide any additional information!

-Mark

LorenzMeier, 2019-02-03:
@bkueng This sounds like we should follow up in detail.

mcsauder, 2019-02-08:
@bkeung, the test was functional after this commit: 3859bbb3b0abdfa1314bb070e4d4aa4aec5d883b

@davids5, somewhere in the next stack of commits it is broken, but many of the commits will not build for me so I cannot determine easily where the breakage occurs.

I do not know if the breakage is in the unit test or in the code.  I hope this helps.

davids5, 2019-02-08:
@mcsauder Fixed in https://github.com/PX4/Firmware/pull/11414

closed 2019-02-08 by dagar