PX4/PX4-Autopilot #11364: VEHICLE_CMD_DO_REPOSITION scaling check invalid for negative lat/lon

blutack, 2019-02-01:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

julianoes, 2019-07-15:
Oh wow, thanks for raising this!

blutack, 2019-07-15:
No problem @julianoes and thanks for unmarking it, it's not major but one of those things that can bite.

If you let me know the preferred option (a, b, c) for resolving I'm happy to open a pull request.

Also reading back my original report, sanity check is probably the wrong phrase to use!

julianoes, 2019-07-15:
Right, so a) would be the first quick fix I'd say.

Then, I would argue that DO_REPOSITION should be documented on the MAVLink level that it always needs to be a COMMAND_INT, and then we could start to reject COMMAND_LONG.

hamishwillee, 2019-07-15:
@julianoes Don't let this stop you but ...

> Then, I would argue that DO_REPOSITION should be documented on the MAVLink level that it always needs to be a COMMAND_INT, and then we could start to reject COMMAND_LONG.

FWIW I'd reject COMMAND_LONG for this message straight away. 

For the mavlink docs we can easily enough doc individual cases, but I think that people are unlikely to check this. I'd rather look at the LONG vs INT problem holistically - perhaps get rid of the COMMAND_LONG altogether. Maybe use markup and tooling to reject use of the wrong message at the mavgen library level. 

There is an outstanding issue on this topic: https://github.com/mavlink/mavlink/issues/1090

garfieldG, 2019-09-03:
Hi, 

wanted to know is there something new regarding this issue? 

regarding the use of reposition only with command_int - I think it should be rethink , as for now it's possible to send a new reposition command with no lat and lon (NAN), but it's possible only with command_long. I think it's something that should be reconsider so that we don't loose this ability.

hamishwillee, 2019-12-03:
@garfieldG Looks like Julian suggested option a is best (https://github.com/PX4/Firmware/issues/11364#issuecomment-511469527) - so PR?

BazookaJoe1900, 2019-12-18:
this issue

closed 2019-12-18 by BazookaJoe1900