PX4/PX4-Autopilot #11380: SITL intermittent failure: [Err] [gazebo_mavlink_interface.cpp:1018] poll timeout

dagar, 2019-02-05:
[image]

Examples
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-11379/1/pipeline

dagar, 2019-02-05:
I've only seen this failure a few times, but so far it's always been in the FW mission test.

dagar, 2019-02-05:
Actual error coming from the sitl_gazebo side (https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L1017-L1019). FYI @TSC21

lamping7, 2019-02-05:
Yeah you can find that it changed in relation to https://github.com/PX4/Firmware/issues/11144. I pointed that out in the other issue tracking this.

ChrisHughes, 2019-02-05:
I'm getting this error following the sending of an invalid or mis-formed bitmask for MAV_FRAME_LOCAL_NED message.

lamping7, 2019-02-05:
Hmm that doesn't seem to follow this issue as we use the same test structure and messages for all vehicles we test in Jenkins. This issue is intermittent with the plane model we use, not the multirotor or vtols tested.

lamping7, 2019-02-06:
I noticed that MAVROS doesn't report `VER` info at startup. So it's likely not connecting to PX4. Maybe another clue to all this.

julianoes, 2019-02-06:
I can't reproduce this on Ubuntu, I ran it a couple of times. @dagar were you able to reproduce this and if so with what configuration? With or without docker? Any specific order that I should do things?

julianoes, 2019-02-06:
I tried delaying `px4` or delaying `gzserver` and it worked for both cases. To me it looks like some kind of race between the PX4 things and MAVROS and what not, at least that's what it looks like with the scrambled log output.

lamping7, 2019-02-06:
> To me it looks like some kind of race between the PX4 things and MAVROS and what not

I don't think it is. Gazebo and MAVROS talk to PX4 separately and they're both not working. Sure MAVROS isn't working, but that's a symptom, not a cause. Why would this only effect the FW vehicle?

julianoes, 2019-02-07:
@lamping7 you're probably right :confused:.

julianoes, 2019-02-07:
From a first look at #11401 I can see that lockstep initializes correctly for the unsuccessful test but then PX4 stops publishing `HIL_ACTUATOR_CONTROLS` messages (or somehow `gzserver` stops to receive them).
I'll have to add more printfs to get closer to this.

dagar, 2019-02-11:
Not fixed. https://github.com/PX4/Firmware/pull/11401

dagar, 2019-02-18:
Possible fix! https://github.com/PX4/Firmware/pull/11485

dagar, 2019-02-18:
Fixed by https://github.com/PX4/Firmware/pull/11485

closed 2019-02-18 by dagar