PX4/PX4-Autopilot #11386: Issues with omnibus target not booting

r1b4z01d, 2019-02-05:
**Describe the bug**
When I flash an omnibusf4 board I can not get the board to boot or connect to GQC. I have tried multiple Hobbywing XRotor Flight Controllers, the old and v2. I have tried multiple SDcards. I have used an SD card that is working in a Pixracer. I have compiled the FW on multiple computers. The only issue is see while building is, I can not upload because the FW is too large for the board. I removed some unneeded drivers to make it fit. PX4Flow, GPS, frsky_telemetry were the last drivers I tried disabling. I have tried V1.8.0 V1.8.2 and V1.9. Is anyone else seeing this? When I plug into QGC it never connects and I get an error to let the system fully boot and restart QGC. 

**To Reproduce**
Steps to reproduce the behavior:
1. make omnibus_f4sd_default
2. make omnibus_f4sd_default upload
3. reboot board
4. plug in to computer and open QGC

**Expected behavior**
          I expect the board to boot and connect to QGC.

**Log Files and Screenshots**
           I do not have any logs because the board does not boot.

bkueng, 2019-02-07:
Hi
What bootloader did you flash? There was a breaking change in https://github.com/PX4/Bootloader/pull/124. Can you try flashing latest master of both the bootloader and the firmware?

r1b4z01d, 2019-02-12:
I have tried the latest bootloader on two HobbyWing boards with zero success. They flash fine but GQC does not connect. See the log here:
https://i.imgur.com/wimiZiO.png

bkueng, 2019-02-13:
Can you provide the exact instructions on how you flashed everything? What's the exact error that you get?
You not only need the latest bootloader, but also master of the Firmware.

closed 2019-07-08 by stale[bot]