PX4/PX4-Autopilot #11393: sitl with gazebo: error opening serial device

jlkjiji, 2019-02-06:
Hi.
I switched to stable branch and run "make posix_sitl_default gazebo" but got error message as "error opening serial device".
[image]
And then followed as:
[image]
While the hil simulation with gazebo is working properly.
I have tried connecting my pixhawk then "make posix_sitl_default gazebo" and it worked just like hil mode.
Can anybody help?

closed 2019-02-16 by jlkjiji

cesar-source, 2021-05-06:
Hi @Alaric-F 

I'm having the same issue... were you able to resolve your problem?

julianoes, 2021-05-07:
Could it be that you still have `serialEnabled` in your iris.sdf file after HITL?

cesar-source, 2021-05-07:
Ah yes, I missed that. Thanks for that!

hsrfeyzullah, 2022-01-27:
How exactly did you solve the problem?

hsrfeyzullah, 2022-01-27:
How exactly did you solve the problem?@cesar-source

cesar-source, 2022-01-27:
@hsrfeyzullah under the `mavlink_interface` plugin in the .sdf file, there should be a `serialEnabled` tag. I set this to 0 and it worked!