PX4/PX4-Autopilot #11402: Failsafe triggering in SITL

ChrisHughes, 2019-02-06:
During SITL test flight using 1.9.0 firmware, Drone executes setpoint command in offboard and then will show the following message
```
Failsafe: Descend with land-speed
```
before landing. I have tried several ways to isolate the problem, and have now started stripping out all custom code & parameters to just run a single MAVROS demo script that should work, and am still getting the error.

Drone is Iris with optical flow.

Firmware tested has been 1.8.2 and 1.9.0 beta.

Using ROS Melodic and Gazebo 9 on Ubuntu 18.04 LTS.

This also happens when eliminating Mavros from the equation and sending standard mavlink offboard messages normally. While I have no PX4 logging, the behaviour of the the drone in real testflights suggests that it encountered the same bug.

There are three points in mc_pos_control_main.cpp that call failsafe, this is the first instance on line 678 

https://github.com/PX4/Firmware/blob/504372f551d37d64141397c9add67c9d35b2338e/src/modules/mc_pos_control/mc_pos_control_main.cpp#L678

This failsafe appears when FlightTasks.cpp's update command returns true. This is triggered by having flight tasks that are still active after calling the internal _updateCommand() function. At this point I decided it's best to post a bug report as troubleshooting what task remains active and why is getting really opaque, perhaps someone on here knows what is causing this issue? I am just trying to get a very simple flight demo together in SITL for now and this is a major blocking issue.

julianoes, 2019-07-15:
Thanks for the issue and sorry for the late response.

@MaEtUgR could you have a quick look?

MaEtUgR, 2020-02-19:
Sorry for the delay. "Failsafe: Descend with land-speed" indicates that either the flight task failed executing which if you're using FlightTaskOffboard can happen in the following 4 cases:
https://github.com/PX4/Firmware/blob/01b280f7ae1e0b222ed0ea91062392e43b8d2c85/src/lib/flight_tasks/tasks/Offboard/FlightTaskOffboard.cpp#L75
https://github.com/PX4/Firmware/blob/01b280f7ae1e0b222ed0ea91062392e43b8d2c85/src/lib/flight_tasks/tasks/Offboard/FlightTaskOffboard.cpp#L176
https://github.com/PX4/Firmware/blob/01b280f7ae1e0b222ed0ea91062392e43b8d2c85/src/lib/flight_tasks/tasks/Offboard/FlightTaskOffboard.cpp#L181
https://github.com/PX4/Firmware/blob/01b280f7ae1e0b222ed0ea91062392e43b8d2c85/src/lib/flight_tasks/tasks/Offboard/FlightTaskOffboard.cpp#L212
or the setpoint output of the flight task is invalid and cannot be executed e.g. some axis is missing.
I would guess you are running in one of the four listed cases but am not familiar enough with the offboard interface to understand why. Also there's no log where the issue is visible which makes it harder for anyone to spot the issue.

julianoes, 2020-05-20:
Closing because there was no follow up.

closed 2020-05-20 by julianoes