PX4/PX4-Autopilot #11408: Issue with Master_vehicle crash

Tony3dr, 2019-02-07:
Today we tested Master v1.9.0 (beta) (7f3b1700) with the Pixracer, Pixhawk 4 mini, Pixhawk 4. Created mission.  The vehicle armed good, took off. At 30 meters the vehicle started spinning and came down to the ground. Resulted in a crash. Same behavior for all 3 platforms. 

Steps to reproduce the behavior:
1. Uploaded mission
2. Drone switched on
3. Vehicle armed
4. Took off
5. Elevated to 30 meters
6. vehicle started spinning out of control
7. The vehicle came down and crashed

Pixhawk racer:
https://review.px4.io/plot_app?log=4f0c2692-5f27-4d83-bfbd-765bf48a989e

Pixhawk 4 Mini:
https://review.px4.io/plot_app?log=53fa724b-906d-4e93-87fe-5a4f4c2a50b8

Pixhawk 4:
https://review.px4.io/plot_app?log=69f42b00-6259-4a97-96c1-bd8f06b4d29d

@dagar

mhkabir, 2019-02-07:
Potentially https://github.com/PX4/Firmware/pull/11376

dagar, 2019-02-07:
Wow, ouch. Looking.

dagar, 2019-02-08:
In both cases (fmu-v4 and fmu-v5) the pc points to math::max (mathlib/math/Limits.hpp:72) called from navigator MissionBlock::get_time_inside().

https://github.com/PX4/Firmware/blob/c3d6a042e6332792fe85a1121c08fefa7dbfcb43/src/modules/navigator/mission_block.cpp#L480-L491

dagar, 2019-02-08:
This is easy enough to fix once off, but I'm looking for a most exhaustive solution (compiler warning perhaps).

dagar, 2019-02-08:
Newer gcc has `-Waddress-of-packed-member`.

dagar, 2019-02-08:
WIP fixes in https://github.com/PX4/Firmware/pull/11411

dagar, 2019-02-08:
We need to do a quick pass of packed structure usage looking for anything that might take the address of a  member, create a reference, or pass it into a function that does.

This includes navigator (mission_item), mavlink (anything from the mavlink spec), and a few drivers.

commit c1732cc9 (2019-02-08): WIP: mathlib Limits don't bother using references

 - cleanup Limits.hpp
 - add -Waddress-of-packed-member where available
 - fixes #11408
commit bd48a332 (2019-02-08): WIP: mathlib Limits don't bother using references

 - cleanup Limits.hpp
 - add -Waddress-of-packed-member where available
 - fixes #11408
commit 3c987a3f (2019-02-08): WIP: mathlib Limits don't bother using references

 - cleanup Limits.hpp
 - add -Waddress-of-packed-member where available
 - fixes #11408
commit 2c40cbcc (2019-02-08): WIP: mathlib Limits don't bother using references

 - cleanup Limits.hpp
 - add -Waddress-of-packed-member where available
 - fixes #11408
closed 2019-02-08 by dagar