PX4/PX4-Autopilot #11410: Compilation error for px4_fmu-v5_default: incompatible cast

Hoeze, 2019-02-08:
**Describe the bug**
When running `make px4_fmu-v4_default` I get the following error:
```
wdog/wd_start.c: In function 'wd_expiration':
wdog/wd_start.c:151:20: error: cast between incompatible function types from 'wdentry_t' {aka 'void (*)(int,  unsigned int,  ...)'} to 'void (*)(int)' [-Werror=cast-function-type]
                 (*((wdentry0_t)(wdog->func)))(0);
                    ^
```

**To Reproduce**
Running the following on my Arch linux:
```
git clone git@github.com:PX4/Firmware.git
cd Firmware
make submodulesclean
make px4_fmu-v4_default
```

I tried the same with v1.9.0-beta1 (`git checkout v1.9.0-beta1`), with the same result.

**Expected behavior**
It should compile.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

dagar, 2019-02-08:
Currently we're using GCC 7 in the CI system and as the only supported compiler version.

You'll need to either fix these small issues (or disable the warnings) or switch to the GCC version in the documentation. We'll need to fix these warnings eventually when we upgrade the toolchain, but we only do that as needed (it's a huge testing effort).

dagar, 2019-02-08:
I'll try to fix the cmake warnings (they go away if you upgrade cmake).

Hoeze, 2019-02-08:
@dagar thanks for looking into it.
I'm using `cmake v3.13.4`.
The only newer version of cmake available would be a release candidate.
Are you sure, you're using release candidate versions of cmake?

**However, my problem aren't the cmake warnings:**
My problem is that `wdog/wd_start.c` does not compile.

The compiler I'm using is `arm-none-eabi-gcc v8.2.0`.
Since this version is also newer than gcc v7, the code should compile anyways, shouldn't it?

Hoeze, 2019-02-09:
OK, I tried compiling with gcc 7.3 and this produced a working binary.
So, there is really a problem with gcc 8.2...

closed 2019-07-08 by stale[bot]

julianoes, 2019-09-24:
Same as #12814.