PX4/PX4-Autopilot #11420: MC - Loops maneuvers when MPC_XY_CRUISE > MPC_XY_VEL_MAX

BazookaJoe1900, 2019-02-09:
Reducing the MPC_XY_VEL_MAX to lower value then MPC_XY_CRUISE make the drone to fly in very strange way, it makes "loops" and fly backwards every few seconds.

This was tested on simulator only.
https://logs.px4.io/plot_app?log=14e8cf1c-2fca-4cf8-ba0d-26ea8c5dd5d1
To reproduce:
•	Run simulator, with default parameters
•	takeoff and fly-to remote location
•	everything normal
•	set MPC_XY_VEL_MAX 2
•	you will see that every ~10 second the drone fly back, do some strange loop
•	set MPC_XY_ CRUISE 2
•	the drone fly slow, but ok
[image]

dagar, 2019-02-11:
@bresch @Stifael any input here?

bresch, 2019-02-11:
Ouch, we should force cruise < max

BazookaJoe1900, 2019-03-07:
@bresch how can you 'force' that? is it ok by you to change the parameter flash value if it doesn't apply the rule? or just whereever it being get() do the constrain?

bresch, 2019-03-07:
Usually we constrain the parameter when we read using `.get()` and change the parameter using a `.set(val)`.

BazookaJoe1900, 2019-03-07:
@bresch for MPC_XY_CRUISE, there is no single point where we read the parameter and can modify it, it uses the new px4_params defined that I am not sure  where I can rightlly fix that.

bresch, 2019-06-25:
Don't close that

BazookaJoe1900, 2019-06-29:
I am not sure what fixed this behavior but it doesn't happens on current master enymore.
@bresch you can check this too and close

bresch, 2019-07-06:
Fixed by #12404

closed 2019-07-06 by bresch