PX4/PX4-Autopilot #11433: LPE Q Estimator external heading fusion not handling lack of external data

ArkadiuszNiemiec, 2019-02-11:
I am using an LPE with an external vision system and yaw information fused in Q Estimator. It works pretty well even without a mag sensor (this currently does not work in EKF2). The problem is that my external estimator can have up to a few seconds pauses in which no new yaw data is published.
I have found out that if the last yaw position is e.g. x radian the Q Estimator is still converging to this number even if gyro has correctly reported 90-degree turn.

Steps to reproduce the behavior:
1. PX4 with LPE estimator, external vision estimator and external vision yaw fused in Q Estimator.
2. Turn off mag sensor.
3. Let the `ATTITUDE.yaw` converge to `VISION_POSITION_ESTIMATE.yaw`.
4. Turn off external vision system.
4. Rotate the autopilot 90 degrees.
5. Watch `ATTITUDE.yaw` converge back to the last data reported by the external vision system.

Expected behavior:
The Q estimator should turn off external vision data if last message is too old.

This screenshot should explain it:
[image]

I have a fix ready and will create a PR after tests.

TSC21, 2019-02-11:
@ArkadiuszNiemiec a PR with a fix is welcomed, yes. Thanks

ArkadiuszNiemiec, 2019-02-11:
@TSC21 I have tested the #11436 in flight and it works perfectly. Here's the screenshot of the data generated in the same way as the previous one:
[image]

closed 2019-02-15 by ArkadiuszNiemiec