PX4/PX4-Autopilot #11441: Add support for MAV_CMD_REQUEST_MESSAGE

hamishwillee, 2019-02-11:
Added in https://github.com/mavlink/mavlink/pull/1079, `MAV_CMD_REQUEST_MESSAGE` is a one-shot version of MAV_CMD_SET_MESSAGE_INTERVAL that you can use to ask for a single response.

Action is to add implementation for MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES, MAV_CMD_REQUEST_CAMERA_INFORMATION, MAV_CMD_REQUEST_CAMERA_SETTINGS, MAV_CMD_REQUEST_STORAGE_INFORMATION.

Request should return the usual set of MAV_RESULTS, with MAV_RESULT_UNSUPPORTED if particular message is not handled.

@dogmaphobic FYI, because many of those messages are yours.

hamishwillee, 2019-06-27:
Still desirable.

hamishwillee, 2019-08-01:
Note, this duplicates https://github.com/mavlink/mavlink/pull/1180

hamishwillee, 2019-11-03:
We're going to :-)

dayjaby, 2020-05-03:
> Request should return the usual set of MAV_RESULTS, with MAV_RESULT_UNSUPPORTED if particular message is not handled.

are you sure about UNSUPPORTED? MAV_CMD_REQUEST_MESSAGE *is* (or rather will be) supported. I think DENIED makes more sense according to the current wording in the mavlink documentation.

hamishwillee, 2020-05-03:
@dayjaby You're correct: `MAV_CMD_REQUEST_MESSAGE` itself is supported, so returning [MAV_RESULT_UNSUPPORTED](https://mavlink.io/en/messages/common.html#MAV_RESULT_UNSUPPORTED) is definitely incorrect (denied is correct).

My original thinking here was that we'd use unsupported to indicate that streaming a particular message is unsupported vs just that they might have additional params that need to be set. But creating an inconsistency in API use like this is not smart. Thanks for pointing this out.

hamishwillee, 2020-08-03:
This has been done. Since not every message can be streamed/emitted, the default behaviour is to return "denied" for messages for which a stream base class has not been created. Creating a base class is trivially easy though, if needed.

closed 2020-08-03 by hamishwillee