PX4/PX4-Autopilot #11443: Mavros not able to communicate with PX4 SITL after the latest upgrades.

SnShine, 2019-02-12:
This whole setup used to work a month ago: https://dev.px4.io/en/simulation/ros_interface.html.

Used `roslaunch px4 mavros_posix_sitl.launch` to launch mavros, px4 sitl, gazebo.

I pulled latest changes from Firmware and all submodules and since then mavros couldn't connect to PX4 SITL. And `rostopic echo /mavros/state` shows connection is false.

Anyone else facing a similar issue and how to resolve this?

TobiMiller, 2019-02-12:
yep. same here.

TSC21, 2019-02-12:
How are you setting up your workspace? For this to work properly, put the Firmware folder inside the a catkin_ws src folder and create a symlink to the sitl_gazebo to a folder at the same level (meaning Firmware and sitl_gazebo should be at the same level in the src folder). Then catkin build tbe workspace and source its setup.bash. everything should work as expected.

TobiMiller, 2019-02-12:
Hey. 
Found the problem. i've written a setup.bash which had the old "posix_sitl_default" path in it:
`#!/bin/bash
cd ~/src/Firmware

source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/**px4_sitl_default**
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
cd ~`

SnShine, 2019-02-13:
Hello @TSC21, I'm git cloning recursive Firmware at ~/ and then doing 

```cd <Firmware_clone>  
make px4_sitl_default gazebo  
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default  
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)  
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo  
roslaunch px4 posix_sitl.launch```

The same worked a month ago. Yesterday, just did `make px4_sitl_default gazebo` once again and it upgraded all the submodules and stopped working since then.

I'd don't have the access to my system RN. Will try what you've suggested and update in a couple of days.

SnShine, 2019-02-13:
Hey @TobiMiller, I didn't get you. Did you solve your problem? I've done the same thing you've mentioned a month ago and doing the same now. It worked then but having problems now.

And @TSC21, I'm following this tutorial https://dev.px4.io/en/simulation/ros_interface.html. And they haven't done anything related to `catkin build`.

TSC21, 2019-02-13:
The tutorial assumes you have to setup your paths manually to make it work. If you build all inside a catkin_ws, you will never have to worry on manually exporting paths.

TobiMiller, 2019-02-13:
@TSC21 correct. this was not the case a long time ago :P

SnShine, 2019-02-13:
@TobiMiller I have installed it via apt.

SnShine, 2019-02-15:
Couldn't figure out what exactly is the issue but solved it setting up my environment again following this: https://dev.px4.io/en/setup/dev_env_linux.html#gazebo-with-ros.

closed 2019-02-15 by SnShine

Dieptranivsr, 2020-09-22:
You may have switched to HITL mode but not SITL.
You should check line code in file iris.sdf
```sdf
<serialEnabled>0</serialEnabled>
      <!--<serialEnabled>1</serialEnabled>-->
      <serialDevice>/dev/ttyACM0</serialDevice>
      <baudRate>921600</baudRate>
      <qgc_addr>INADDR_ANY</qgc_addr>
      <qgc_udp_port>14550</qgc_udp_port>
      <hil_mode>0</hil_mode>
      <!--<hil_mode>1</hil_mode>-->                        
```