PX4/PX4-Autopilot #11445: [Feature] Handle avoidance setpoints correctly regarding smoothing

AuterionWrikeBot, 2019-02-12:
@Mathieu Bresciani  could you please fill in the details what this would entail

jkflying, 2019-02-13:
@mrivi FYI

bresch, 2019-02-14:
Update after discussion with @mrivi : 
- We need to have avoidance available in manual position and auto modes.
- We want to have the overall same flight behavior when the drone flies a regular mission (no obstacles) with and without avoidance:
   -  *Solution*: Smoothing has to be done in the FlightTask or stop publishing avoidance when there are no obstacles.
- To avoid discontinuities in the setpoints when the avoidance starts and stop publishing, the FlightTask has to be aware of the avoidance setpoint 
   - *Solution*: Read avoidance setpoints in the FlightTasks
- If avoidance wants to stop quickly the drone, we don't want to add delay because of the smoothing 
   - *Solution*: Use a second set of jerk/acceleration constraints when the avoidance publishes a setpoint to allow fast reaction.

MaEtUgR, 2019-02-28:
Sidenote: it's confusing when the @AuterionWrikeBot creates issues. Could at least be in the description who wrote it?

> we need to read and use those setpoints directly in the FlightTask

yes ok, but we should keep it contained in a library like @mrivi started out in #11561 because if it's modular in this way it's better maintainable.

> We need to have avoidance available in manual position and auto modes.

Does the avoidance always react the same way? I expected a different behavior when flying manually compared to a mission.

mrivi, 2019-02-28:
> We need to have avoidance available in manual position and auto modes.

> Does the avoidance always react the same way? I expected a different behavior when flying manually compared to a mission.

PX4 avoidance can be run only in auto modes. However the current interface supports pos control modes as well (even if they are not enabled by default).

jkflying, 2019-03-01:
We have collision prevention available in manual position control, using the laser scanner message.

MaEtUgR, 2019-03-01:
> We have collision prevention available in manual position control, using the laser scanner message.

Ok, that's what I expected, the implementation here https://github.com/PX4/Firmware/tree/6c3e79f3614f0060d5998c39be358512101252ff/src/lib/CollisionPrevention is used with distance data from the external module in manual position flight.

mrivi, 2019-03-02:
The PX4 collision prevention doesn't uses the `vehicle_trajectory_waypoint` interface these changes don't have an impact on that feature. 
However I am aware of other obstacle avoidance system integrated with PX4 that can be used in position control mode. Therefore the current master implementation is thought for pos control mode as well even if it's not enabled by default upstream since PX4 avoidance doesn't support it.

jkflying, 2019-07-03:
This is fixed now.

closed 2019-07-03 by jkflying