PX4/PX4-Autopilot #11450: I tried to compile the 1.8 version program.Use this command “make px4fmu-v5_default”  。I need your help. It reminds me of this information.thanks！

1xiao-xiong, 2019-02-13:
[100%] Linking CXX executable ../../nuttx_px4fmu-v5_default.elf
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_fpclassify.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_fpclassify.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_cos.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_cos.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_sin.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-s_sin.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_atan.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_atan.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_cos.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_cos.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fabs.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fabs.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fmax.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fmax.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fmin.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fmin.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fpclassify.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_fpclassify.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_sin.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_sin.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_tan.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-sf_tan.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_acos.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_acos.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_asin.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_asin.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_atan2.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_atan2.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_sqrt.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-w_sqrt.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_acos.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_acos.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_asin.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_asin.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_atan2.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_atan2.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_exp.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_exp.o)
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v5_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_fmod.o) does not
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/libm.a(lib_a-wf_fmod.o)
[... cut ...]
platforms/nuttx/CMakeFiles/nuttx_px4fmu-v5_default.elf.dir/build.make:268: recipe for target 'nuttx_px4fmu-v5_default.elf' failed
make[3]: *** [nuttx_px4fmu-v5_default.elf] Error 1
CMakeFiles/Makefile2:9361: recipe for target 'platforms/nuttx/CMakeFiles/nuttx_px4fmu-v5_default.elf.dir/all' failed
make[2]: *** [platforms/nuttx/CMakeFiles/nuttx_px4fmu-v5_default.elf.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:153: recipe for target 'px4fmu-v5_default' failed
make: *** [px4fmu-v5_default] Error 2
root@ubuntu:/home/pix/code/Firmware#

mhkabir, 2019-02-13:
Please use the officially supported GCC 7.2.1 arm toolchain. You are running 4.9.3.

closed 2019-02-13 by mhkabir

mormank, 2019-07-03:
@mhkabir any ideas for people who are using older versions because for 32-bit linux? Namely, raspbian?