PX4/PX4-Autopilot #11451: Pixhawk 4 IRLock test Freeze up and bricking

JagerBombastic, 2019-02-13:
Hello,
I am running  irlock on a pixhawk 4 and I have found that with both v1.8.2 and v1.9beta1 when the irlock test is run from the mavlink cli and you allow a largish number of results to come though (<10) it has a 50/50 chance of either freezing and rebooting the FC or soft bricking it which then requires re-flashing the firmware to recover. 

I am running a Lidar Lite v3 on the same i2c bus but i'm inclined to think that's not relevant.

dagar, 2019-02-13:
What do you mean "allow a largish number of results to come though"? How are you controlling this?

JagerBombastic, 2019-02-14:
so when the 'irlock test' is run it will print lines of information about what the sensor is seeing (x,y data ect...) and It will only send data through when the sensor can see the beacon. 
If i cover the sensor then briefly allow it to view the sensor it will print a few lines and doesn't crash. 
If i allow it constantly view the beacon the test will produce 10-20 lines and then it will freeze and the either reboot or soft-brick the pixhawk.

narew, 2019-02-17:
It looks We have the similar problem but mine is on Pixhawk 2.1: 
https://github.com/PX4/Firmware/issues/11477

JagerBombastic, 2019-02-17:
I can at least say it only the test that causes this. In the field irlock works fine.

bys1123, 2019-02-19:
If not use IRLock, will it freeze up when you use the FC on 1.8.2?

ndepal, 2019-03-19:
Might be a stack overflow from the many print statements.
You could try commenting out the print statement of the test routine to verify.

closed 2019-07-08 by stale[bot]

dvornikov-aa, 2019-11-18:
I've experienced freeze and reboot problems. Can't confirm the bricking. According to the debug port output before the crash there is a low on stack warning.