PX4/PX4-Autopilot #11452: Takeoff not responding properly in position control

bresch, 2019-02-13:
@MaEtUgR I create this Issue to keep track of what we discussed on slack.

**Describe the bug**
While armed in position control with the throttle stick low, the flight task generates a negative setpoint. When now the user moves the throttle stick above 50%, the setpoint has to go positive before triggering the smooth takeoff. However, with the dynamics of the smooth flight tasks, this takes and depends of some parameters such as the max down velocity, vertical acceleration and jerk limitations.
I see two ways to solve that:
1. Make the flight task aware of the current state (the flight task will then not output a downward sertpint while on ground)
2. Modify the constraint in mc_pos_control and feet it back to the flight task (I tried that on a branch)
The problem is that both solutions introduce a feedback to the flight task and I know that @stifael wanted to avoid that. Do you have an other solution? 

**To Reproduce**
In position control mode (using a smoothing FlightTask with low vertical acceleration and jerk)
Set the throttle stick to low
Arm and wait a few seconds
Raise the stick above 50%, it will take some time before taking off

To avoid this issue :
- Use a spring-loaded throttle stick or center the throttle stick before arming and use an arm switch
- Arm and directly center the throttle stick

**Expected behavior**
The drone should takeoff as soon as the throttle stick is above 50%

**Log Files and Screenshots**
[image]

**Proposed solution**
I tried the 2nd solution proposed in the description and limit the downward velocity constraint while the vehicle is on ground, here is the result:
[image]

bresch, 2019-02-13:
Note that having feedback to the FlightTask will make #10659 easy to fix.

Stifael, 2019-02-13:
I mean it really depends what kind of feedback it is. For instance, if the vehicle is landed and not in smooth-takeoff, the flighttask reactivate itself (https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L736). During this reactivation, the setpoint-smoothing could be turned off

Stifael, 2019-02-13:
Just guessing: you probably could reset the smoothing in [here](https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/ManualPositionSmoothVel/FlightTaskManualPositionSmoothVel.cpp#L41)

bresch, 2019-02-13:
@Stifael The problem with the reactivate strategy is that I disabled it for the SmoothVel FlightTask because it can't reach the required velocity setpoint to trigger smooth takeoff in one `dt`. Can I maybe add a condition for smooth takeoff on the jerk setpoint?

PR #11460 (2019-02-25): smooth takeoff - Support smooth takeoff triggered by jerk setpoint
fixes #11452 

**Test data / coverage**
Tested in SITL
[image]

**Describe problem solved by the proposed pull request**
A (hard to predict) delay before takeoff was present on all the non spring-loaded remotes while using the `SmoothVelocity` position sub mode. This PR enables a full reset of the `ManualSmoothVel` FlightTask by adding a jerk-based check to trigger "smooth takeoff" in `mc_pos_control_main`.
Takeoff is now triggered as soon as the throttle stick is above 50%.

**Describe possible alternatives**
Setting the `vel_ [... cut]
closed 2019-02-25 by bkueng