PX4/PX4-Autopilot #11458: Problem in spawning multiple drones in Gazebo

DharnaNar, 2019-02-14:
I am not able to launch multi-uav-mavros-sitl with the latest firmware.

Here are the commands I am using:
cd catkin_ws/src/Firmware
git submodule update --init --recursive
make px4_sitl_default
make px4_sitl_default sitl_gazebo

source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo

roslaunch px4 multi_uav_mavros_sitl.launch

There is no error that shows up, but gazebo doesn't show 3 drones . It shows nothing. Neither are the drones connected over udp.

If I try to launch single_vehicle_spawn launch file , even that doesn't get launched.

pravardhanreddy, 2019-02-15:
Same issue. Have been trying to fix it for over a month. It was working on ubuntu 16 Ros kinetic few months back. @devs please fix this.

pravardhanreddy, 2019-02-17:
@k2shah can you please look into this and get this resolved. Similar sort of issues happened before, as I remember you and @bkueng were able to resolve this in the past.

k2shah, 2019-02-17:
hi @pravardhanreddy  
 I'm not able to set aside time to handle this at the current moment. 
if you don't need the current firmware I mention here https://github.com/PX4/Firmware/issues/10430 that you can clone our labs fork https://github.com/StanfordMSL/Firmware/tree/multi-quad

last i checked it was working on 16/kinetic and 18/lunar

I will look into this as soon as some other tasks of mine have been resolved.

DharnaNar, 2019-02-17:
I had worked on the same script 2 months back and it was working properly. I don't have a backup of that PX4 firmware. Need ti find a solution to this asap.

k2shah, 2019-02-17:
if you clone our fork of the px4 repo at the multi-quad branch it should roll back the changes made to the udp port changes that were made

pravardhanreddy, 2019-02-17:
@k2shah what arguments should i use for make?

k2shah, 2019-02-17:
you should just be able to pull that branch form our fork and make as normal (after the git update)

`git submodule update --init --recursive
make px4_sitl_default
make px4_sitl_default sitl_gazebo`

pravardhanreddy, 2019-02-17:
I get the following error
_Makefile:422: *** "px4_sitl_default cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop._

k2shah, 2019-02-17:
it might just be `make siti_default`

k2shah, 2019-02-17:
hmm i don't remember the exact make line, check the simulation documentation the makes you need to run should be there

pravardhanreddy, 2019-02-17:
Ok so there is some progress. I used the following commands for make
`make posix_sitl_default`
`make posix_sitl_deafult sitl_gazebo`
First I tested for snigle vehicle simulation
`roslaunch px4 mavros_posix_sitl.launch`
The drone got spawned in gazebo with its motors spinning slowly for some reason, all the mavros topics were also listed under
`rostopic list`
but they are not publishing anything, when i do 
`rostopoic echo /mavros/imu/data`
i get the following warning
_WARNING: no messages received and simulated time is active. Is /clock being published?_
I checked /clock it was publishing.
Seems like mavros is not able to connect to the drone.
I am testing on Ubuntu 18.04/melodic

pravardhanreddy, 2019-02-17:
same issue with multi vehicle
All the three vehicles are spawned, also with their motors spinning slowly (don't know why that is happening).
All the mavros topics are also listed under their respective namespaces but they are not publishing any data.

k2shah, 2019-02-17:
are you on 18? can you try a machine with 16 on it? I can try to clone and build when i have access to a linux machine

DharnaNar, 2019-02-17:
i am working on ubuntu16. I am also facing the same issue with https://github.com/StanfordMSL/Firmware/tree/multi-quad , motors are spinning slowly on in multi_uav launch. Also these drones are not getting connected over udp to QGC. Same is the the case for mavros_posix_sitl.launch.

k2shah, 2019-02-17:
Can you pipe the output of the launch file to a txt and send that to me/post it here

DharnaNar, 2019-02-17:
in the latest px4/Firmware repo, the only problem is with multi-uav launch. px4_sitl is not getting connected to simulator. Neither it is getting connected to QGC. 
While roslaunch mavros_posix_sitl.launch works absolutely perfect and i am also able to control the drone using rosrun px4 mavros_offboard_posctl_test.py

k2shah, 2019-02-17:
That makes sense because the port were changed from udp to tcp. There's probably some gazebo <=> px4 port miss match happening

DharnaNar, 2019-02-17:
'source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 multi_uav_mavros_sitl.launch'

these are the set of commands 
and the output is

started roslaunch server http://dharna-Latitude-E6420:41105/

SUMMARY
========

CLEAR PARAMETERS
 * /uav1/mavros/
 * /uav2/mavros/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uav1/mavros/cmd/use_comp_id_system_control: False
 * /uav1/mavros/conn/heartbeat_rate: 1.0
 * /uav1/mavros/conn/system_time_rate: 1.0
 * /uav1/mavros/conn/timeout: 10.0
 * /uav1/mavros/conn/timesync_rate: 10.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/laser_1_sub/id: 3
 * /uav1/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /uav1/mavros/distance_sensor/lidarlite_pub/id: 1
 * /uav1/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/sonar_1_sub/id: 2
 * /uav1/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /uav1/mavros/fake_gps/eph: 2.0
 * /uav1/mavros/fake_gps/epv: 2.0
 * /uav1/mavros/fake_gps/fix_type: 3
 * /uav1/mavros/fake_gps/geo_origin/alt: 408.0
 * /uav1/mavros/fake_gps/geo_origin/lat: 47.3667
 * /uav1/mavros/fake_gps/geo_origin/lon: 8.55
 * /uav1/mavros/fake_gps/gps_rate: 5.0
 * /uav1/mavros/fake_gps/mocap_transform: True
 * /uav1/mavros/fake_gps/satellites_visible: 5
 * /uav1/mavros/fake_gps/tf/child_frame_id: fix
 * /uav1/mavros/fake_gps/tf/frame_id: map
 * /uav1/mavros/fake_gps/tf/listen: False
 * /uav1/mavros/fake_gps/tf/rate_limit: 10.0
 * /uav1/mavros/fake_gps/tf/send: False
 * /uav1/mavros/fake_gps/use_mocap: True
 * /uav1/mavros/fake_gps/use_vision: False
 * /uav1/mavros/fcu_protocol: v2.0
 * /uav1/mavros/fcu_url: udp://:14541@loca...
 * /uav1/mavros/gcs_url: 
 * /uav1/mavros/global_position/child_frame_id: base_link
 * /uav1/mavros/global_position/frame_id: map
 * /uav1/mavros/global_position/gps_uere: 1.0
[... 421 lines cut ...]
[ INFO] [1550389862.886279885, 0.076000000]: Plugin trajectory loaded
[ INFO] [1550389862.989636803, 0.166000000]: Plugin trajectory initialized
[ INFO] [1550389862.989974770, 0.166000000]: Plugin vfr_hud loaded
[ INFO] [1550389862.994598504, 0.172000000]: Plugin vfr_hud initialized
[ INFO] [1550389862.994684776, 0.172000000]: Plugin vibration blacklisted
[ INFO] [1550389862.994910035, 0.172000000]: Plugin vision_pose_estimate loaded
[uav1/iris_1_spawn-5] process has finished cleanly
log file: /home/dharna/.ros/log/c31b841a-3288-11e9-8209-2477030c6464/uav1-iris_1_spawn-5*.log
[ INFO] [1550389863.147705844, 0.182000000]: Plugin vision_pose_estimate initialized
[ INFO] [1550389863.148091253, 0.182000000]: Plugin vision_speed_estimate loaded
[ INFO] [1550389863.176535334, 0.182000000]: Plugin vision_speed_estimate initialized
[ INFO] [1550389863.176960349, 0.182000000]: Plugin waypoint loaded
[ INFO] [1550389863.197110139, 0.182000000]: Plugin waypoint initialized
[ INFO] [1550389863.197406433, 0.182000000]: Plugin wind_estimation loaded
[ INFO] [1550389863.199601794, 0.182000000]: Plugin wind_estimation initialized
[ INFO] [1550389863.199950168, 0.182000000]: Autostarting mavlink via USB on PX4
[ INFO] [1550389863.200173915, 0.182000000]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1550389863.200355683, 0.182000000]: Built-in MAVLink package version: 2019.1.12
[ INFO] [1550389863.200512045, 0.182000000]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1550389863.200660093, 0.182000000]: MAVROS started. MY ID 1.240, TARGET ID 2.1
[INFO] [1550389863.280613, 0.182000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1550389863.351270329, 0.234000000]: Physics dynamic reconfigure ready.
[ INFO] [1550389863.356624835, 0.244000000]: Physics dynamic reconfigure ready.
[uav2/iris_2_spawn-8] process has finished cleanly
log file: /home/dharna/.ros/log/c31b841a-3288-11e9-8209-2477030c6464/uav2-iris_2_spawn-8*.log

DharnaNar, 2019-02-17:
I've done a fresh setup of Ros and Firmware, tree/multiuav one ...and the drones are up and properly connected to gazebo as well as QGC. 
The only error I am getting now is that sensors fail ,Timeout

DharnaNar, 2019-02-17:
ERROR [sensors] Accel #0 fail:  STALE!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Accel #1 priority: 1

DharnaNar, 2019-02-17:
The problem is solved after  git submodule update --init --recursive
Thank you @k2shah .Kindly update here about the latest firmware port updates

julianoes, 2019-02-18:
@DharnaNar is this resolved then and I can close the issue?

bkueng, 2019-02-18:
Related: https://github.com/PX4/Firmware/pull/11479

DharnaNar, 2019-02-18:
Temporarily this issue is resolved, but the ports issue for multi uav is not yet resolved for the latest firmware.

julianoes, 2019-02-18:
ports issue for?

DharnaNar, 2019-02-18:
Spawning multiple UAVs and connecting them over tcp or udp

pravardhanreddy, 2019-02-18:
It is still not working on Ubuntu 18.04/melodic. With the px4/firmware master, gazebo tries to open up but is stuck there nothing is shown, not even an empty world. The problem seems to be with single_vehicle_spawn.launch. With the StanfordMSL/Firmware/tree/multi-quad, the quads are spawning properly but are not publishing anything on the mavros topics. All the topics are listed when i do rostopic list but no data seems to be published on them (checked using rostopic echo).

DharnaNar, 2019-02-18:
Connecting multiuavs px4_sitl over udp to QGC and Gazebo simulator in the latest firmware

DharnaNar, 2019-02-18:
StanfordMSL/Firmware/tree/multi-quad , quads are spawning and publishing on Ros topics on Ubuntu 16 and I am able to control these individually from py scripts

stmoon, 2019-02-19:
I found there is a port issue in the master version because of the mavlink_tcp_port.
I change and pull request at #11479  with PX4/sitl_gazebo.

It is working on Ubuntu 18.04/gazebo 7

julianoes, 2019-02-19:
https://github.com/PX4/Firmware/pull/11479 and https://github.com/PX4/Firmware/pull/11498 have been merged, this should work now. Please re-open if you still have issues, thanks.

closed 2019-02-19 by julianoes