PX4/PX4-Autopilot #11464: AUX1 pass through from RC channel !!!

teletubbiesarenotforpendingpandas, 2019-02-14:
im trying to get a servo/gimball to be controlled from my rc input.

i have seen this question asked a few times but my problem still persists.
im using 

qgc 3.5
px4 latest stable release 
all calibrated

ive choosen in qgc to pass my channel 7 to aux 1 but i get no output at all.even with a multimeter there is no change.

to test this i setup the camera trigger on aux1 and it did cause a jitter in my gimball therfore aux1 is wokring.

i then tried and installed ardupilot flightstack instead of px4 via qgc and managed to get the gimball working fine and all was setup in five minutes.

so this proves the pins aux1 are working and the pwm signal is moving the gimball up and down.

but i still cant get it working on px4 using the aux1/2 passthrough button.

i manually manipulated PWM_AUX_DIS1 and saw that i do get movement.

i have no idea something so simple could cause days of head bangging.
if somoene could help.

tops4u, 2019-02-19:
Does Channel 7 move in the RC Tab? I have a similar Setup and it works without problems. 

Could you post a Screenshot of the RC Tab of QGC?

Pravin854, 2019-02-19:
[image]
facing similar issue

tops4u, 2019-02-20:
Hi @Pravin854 what Type of Hardware are you using?

Pravin854, 2019-02-20:
Flight controller - Pixhawk px4 (https://docs.px4.io/en/flight_controller/pixhawk.html), transmitter - Taranis Q X7 (https://www.frsky-rc.com/product/taranis-q-x7-2/), receiver - FrSky X4R 3/16CH Telemetry (https://www.frsky-rc.com/x4r/). I have tried reading signals in multimeter and in arduino uno, but its not showing any output.

teletubbiesarenotforpendingpandas, 2019-02-21:
i solved my problem,

it seems that depending on what airframe you have selected in setup px4 already defines some sort of output pin mix.
see here to check what you have:https://dev.px4.io/en/airframes/airframe_reference.html

i originally selected Generic Quadrotor x with mount (e.g. gimbal)  as thats what i had.
i changed and selected Generic Quadrotor x without the gimball and it worked straight away.

in the Generic Quadrotor x mix it clearly states AUX1: feed-through of RC AUX1 channel.

also to confirm aux1 rc passthorugh works without needing to be armed  running px4 firmware or ardupilot.

hope that helps.

Pravin854, 2019-02-21:
I've disabled MNT_DO_STAB, but aux pin is still not working. I am using Generic Hexarotor x geometry.
[image]

joydeepb47, 2019-02-23:
Same issue here with VTOL convergence. Can't get a passthrough in Aux1/2. However, If I enable camera shutter option with generic PWM trigger, It works of any Aux channel as assigned. Then I tried copying a pass.aux.mix file in the sd card for rc passthrough still didnt work either.

closed 2019-02-24 by teletubbiesarenotforpendingpandas

bys1123, 2019-04-06:
Did you try to arm?

joydeepb47, 2019-04-06:
The problem was happening because I was trying do it with the master
version. The problem was solved by compilation of the stable version. Just
modified the airframe, added the mixer and recompiled. Thanks anyways..

bys1123, 2019-04-06:
Thanks for the explain, I'll try it.