PX4/PX4-Autopilot #11467: vtol front transition height drop

lingbuweibu, 2019-02-15:
I use 1.8.1 code. When the VTOL conversion is completed, the throttle drops and the plane drops. The given pitch is positive, but the actual pitch is negative. It takes a while for the pitch to rise again. Does anyone know how to debug it?The log is on
[https://logs.px4.io/plot_app?log=d1482f4d-880f-48fa-aca3-023fddb0eed5](url)
Thanks!

lingbuweibu, 2019-02-15:
[image]

JohnSnowball, 2019-02-16:
I think the main cause to this is the pitch down, as you can see the cruise throttle is about 0.7. Throttle after transition is above this level, and shouldn't make the plane lose height. However, why the actual pitch doesn't follow the pitch setpoint right after front transition is finished? 
[image]
[image]
Check the above figures, both pitch angle and pitch angular rate are separated! @dagar