PX4/PX4-Autopilot #11468: Land failsafe broken without GPS

AuterionWrikeBot, 2019-02-16:


julianoes, 2019-07-11:
@bresch is this still an issue?

julianoes, 2019-10-09:
Closing because there was no follow up.

julianoes, 2019-10-09:
On a second read, let me reconsider this and check it.

julianoes, 2019-10-09:
I can reproduce this in SITL using `gpssim stop` and then disconnecting the joystick in QGC:

```
WARN  [commander] Manual control lost
WARN  [commander] Failsafe enabled: no RC
WARN  [mc_pos_control] Auto activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Previous flight task failed, switching to mode 1
WARN  [mc_pos_control] Auto activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Ctrl activation failed with error: Activation Failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
```

It looks like the problem is that it goes into Land instead of Descend mode. Land mode requires a valid position while descend should only require a baro.

FYI @MaEtUgR

julianoes, 2019-10-09:
`NAVIGATION_STATE_DESCEND` is mapped to the land functionality in navigator:
https://github.com/PX4/Firmware/blob/f9a5c91f8cc5f74ce8d2a0425f7b76d7af65a0a8/src/modules/navigator/navigator_main.cpp#L594-L597

PR #13143 (2019-10-18): FlightTasks: add Descend task to land without GPS
This adds a flight task to catch the case where we want to do an emergency descent without GPS but only a baro.

Previously, this would lead to the navigator land class being called without position estimates which then made the flight tasks fail and react with a flight task failsafe. This however meant that landed was never detected and a couple of confusing error messages.

This applies if NAV_RCL_ACT is set to 3 "land".

Fixes #11468.

Tested in SITL with `iris`, `plane`, and `standard_vtol`.
closed 2019-10-18 by julianoes