PX4/PX4-Autopilot #11470: Throttle PID Attenuation parameter bound check

ktskt, 2019-02-16:
**Describe the bug**
Dear, developers.

During the tests, I think MC_TPA_RATE_P range is also somewhat problematic.
It range is from 0 to 1.0.
But, many other values can be assigned and my testing drone shows totally bad performance in flight even with 0.1 and even lower  (a larger value shows worse performance).

I am talking about not only range check (in program) but also documented ranges in https://dev.px4.io/en/advanced/parameter_reference.html
I think it would be better to put some check or at least revise the its valid ranges. 

Such parameters can be modified whenever PX4 runs via GCS command even during flight with SITL
MC_TPA_RATE_P
Its default is 0 which showed no problem.

**To Reproduce**
With GCS connected to a drone, send param set MC_TPA_RATE_P (e.g., even 0.1)

**Expected behavior**
Oscillating flight or crash on the gorund

**Log Files and Screenshots**
I will share the log or screenshot if you need... It is very easy to reproduce....

**Drone (please complete the following information):**
SITL with IRIS+ quadcopter model

julianoes, 2019-07-10:
Good suggestion. @bkueng do you have an opinion?

bkueng, 2019-07-11:
I don't think this is valid, as you can make the system unstable with a specific set of tuning parameters that work perfectly well on another system. TPA is among those parameters.

closed 2019-07-11 by bkueng