PX4/PX4-Autopilot #11473: Tilting angle parameter range check

ktskt, 2019-02-16:
**Describe the bug**
Dear, developers,

Tilting angle parameter (MPC_TILTMAX_AIR and MPC_TILTMAX_LND) should have a minimum value 
I think it is necessary to put some minimal checks...
For example, if there is wind, it will be drifted (I checked with wind plugin included in sitl_gazebo) or fly with a very low speed (even without wind if their values are 0 or very small). 

**To Reproduce**
param set MPC_TILTMAX_AIR or MPC_TILTMAX_LND 0 or low value.
If value is low, it will be drifted to the direction in which wind blows. (reproducible with simulator)

**Expected behavior**
Drifted or fly with very low speed

**Log Files and Screenshots**
I will prepare it if you want .... but it is very easy to reproduce.

**Drone (please complete the following information):**
SITL with IRIS+ quadcopter model

dagar, 2019-02-18:
@bresch @MaEtUgR what do you think? A conservative minimum could be added at the param metadata level.

bresch, 2019-02-19:
@dagar Yes, we should not have the minimum at zero. We can set `MPC_TILTMAX_AIR` min to 30 degrees and `MPC_TILTMAX_LND` min to 10 degrees. That way the user will be warned if he wants to put a lower value.
@ktskt Would that be fine for you?

ktskt, 2019-02-19:
@bresch Personally, I think setting MPC_TILTMAX_LND min to 20 degrees. I think 10 degrees may not be enough to resist normal but strong wind (e.g., 10m/s which is normal according to wind statistics).
Could I ask you why MPC_TILTMAX_LND should be set to 10 degrees?

MaEtUgR, 2019-03-02:
I have nothing against a minimum there since the value 0 doesn't make much sense. But we should choose the minimum value relatively conservative because for certain applications e.g. indoors a small maximal tilt is desirable. As a reference we successfully use `MPC_TILTMAX_AIR` of 35° and `MPC_MAN_TILT_MAX` of 24° for yaers on the H520.
My conservative suggestion would be:
- 15 degree minimum for `MPC_TILTMAX_AIR`
- 5 degree minimum for `MPC_TILTMAX_LND`
This would prevent the drone from drifting away in a normal use case without any heavy wind.

> Personally, I think setting MPC_TILTMAX_LND min to 20 degrees

That limit is above the default value of 12°. `MPC_TILTMAX_LND` is supposed to limit the tilt only during takeoff and shortly before touching the ground on landing to avoid that the vehicle tips over. We should probably revise how it's used right now because if it's the limit for the whole landing descend that's unwanted.

> I think 10 degrees may not be enough to resist normal but strong wind

That is true. It depends on the weight and air resistance of your vehicle but high winds generally require more tilt to counteract the disturbance. You're always free to configure your vehicle to a value that suits your application and needs. If you expect heavy wind and don't have any constraints in terms of attached camera gimbal I suggest to you to set `MPC_TILTMAX_AIR` to the maximum tilt at which your vehicle is still able to maintain it's altitude. If I'm not mistaken the implementation is even safe when you set the value higher and it will prioritize maintaining altitude.

Why do you want to adjust the minimum PX4 configuration to work in high winds?

ktskt, 2019-03-21:
@MaEtUgR Hello, I proposed a conservative suggestion to prevent some strong wind since strong wind can be blown for a while.
But, if you think 15 and 5 are fine (or 30 and 10 as @bresch mentioned), I am fine. 
However, I believe all people agree that there should be some minimum values.

MaEtUgR, 2019-03-21:
I made a pr with a compromise 20° max 10° while landing. The user can still force other values but it gives him a hint that he should not go too low.

@ktskt
> I think setting MPC_TILTMAX_LND min to 20 degrees
The default for the landing tilt is 12°, we cannot set the minimum above the default value. Would you increase the default to 20° based on your experience? The reason it is lower is that depending on your vehicle it might tip over or touch the ground with the propellers while landing with too much tilt. 20° might still be very reasonable given that enables more wind disturbance rejection and still covers most setup types.

bresch, 2019-03-25:
Closed by https://github.com/PX4/Firmware/pull/11696

closed 2019-03-25 by bresch