PX4/PX4-Autopilot #11477: Irlock stable i2C communication problem

narew, 2019-02-16:
**Describe the bug**
Hello,
I am trying to run Irlock (pixycam with a special filter on the lens + ir beacon) on Pixhawk 2.1 with the PX4 firmware. Unfortunately, irlock does not want to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Change the line 66 in src>drivers>irlock>irlock.cpp from 
#define IRLOCK_I2C_BUS			PX4_I2C_BUS_EXPANSION
to
#define IRLOCK_I2C_BUS			PX4_I2C_BUS_ONBOARD
because for Pixhawk 2.1 onboard I2C is appropriate bus.
2. Compile and upload the firmware (px4_fmu-v3_default) to Pixhawk 2.1. own compilation is necessary because by default the irlock is unsupported. 
The: 
-drivers/irlock
-modules/landing_target_estimator
were added for compilation as were sugessted in: https://docs.px4.io/en/advanced_features/precland.html
Unfortunately, I do not know how to make the system start-up described in the above manual
3. In the QGC thru MavlinkConsole test the range finder which also works on same I2C. It is for check the bus is ok. Used distance sensor is LidarLite and appriopriate command is: ll40ls info, ll40ls test
[image]
4. Next in the console I start the irlock driver using command:
irlock start (before change the bus from I2C EXPANSION to I2C ONBOARD I were used command irlock start -b 2)
4. Check the communication by command: irlock info
5. Make some test measurements by: irlock test
[image]
6. If the beacon is off nothing is happen and autopilot works as before, but if beacon is on then the console return some measurements and then communication with pixhawk 2.1 is reseted for a while or autopilot get restart

**Expected behavior**
The autopilot should be stable and return the local position given in the console from the irlock system. Without the solution of this problem, the precise landing function can not be used.

**Drone (please complete the following information):**
- Describe the type of drone.
Quad X type, with Pixhawk 2.1 and Here+ onboard. Radiomodem - RFD 868MHz and the range finder - LidarLite V3, RC tx Frsky X9E and rx X8R. Ground station on QGC v3.4.4, 

**Additional context**
Irlock with described set were tested on Arducopter stack to eliminate possible hardware defects or connections. Irlock worked fine. So It looks there is a problem with the Irlock driver or I2C communication on PX4 stack for Pixhawk 2.1 hardware.

julianoes, 2019-07-10:
Sorry for the late response and thanks for the issue. I wouldn't know who of the core developers has bandwidth to debug what's going on with the irlock pixy. If you can do so and send a pull request with the fixes, that would be fantastic, otherwise we will just have to leave this issue open to track the issue.

julianoes, 2019-10-09:
FYI @ItsTimmy