PX4/PX4-Autopilot #11478: Takeoff setpoint position not same with current position

cimbar, 2019-02-16:
I faced a situation that the setpoint position X differs from zero in takeoff mode although takeoff was commanded for current position. After landing,i didn't change anything and tried takeoff again and again, everything was okay, the setpoint position x was zero again. This was happened only once with same settings.
Log: https://logs.px4.io/plot_app?log=9c85a265-6922-43c8-bc0a-5645f2f4c9d8

Additionally, i don't use GPS and vision position was used in all flights.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
Sorry for the late response. The takeoff position does not necessarily equal the origin (0,0,0) position. The origin is created where the estimator initializes. Does this answer your question @cimbar ?

cimbar, 2019-12-13:
I have been sending position data before initialization. Tested it with different configuration like sending non zero initial position to px4 as vision odometry from companion computer. Then start the mavros and the first position sent is not zero. Whenever i tried to takeoff with initialized non zero position, the fcu tries to takeoff to local zero. Is there any other way to initialize first position? if i land the device after first takeoff and try takeoff again, there is no problem for the second takeoff. But at the first one it does not takeoff at the current position.
Such flight logs,
https://review.px4.io/plot_app?log=1ea01287-4688-4981-bf4d-67881157609d
https://review.px4.io/plot_app?log=2779d356-7f3e-41a8-979c-37ab39344ef8
https://review.px4.io/plot_app?log=833b216f-b51f-4a99-867b-f12dbecb1d8f
https://review.px4.io/plot_app?log=51d13f1f-07e9-4189-81dd-54afcea2f325

julianoes, 2019-12-16:
@cimbar is there a way in SITL to reproduce this?

cimbar, 2019-12-25:
Not exactly, i have added a ros publisher and transform broadcaster inside of gazebo_vision_plugin to publish vision pose in different base frame, then i have used static transform publisher to shift that base.  After that, i used mavros transform listener for vision position estimate. I couldn't run this properly with EKF in SITL and when i used LPE, the initial local position comes zero as expected although the vision position initialized as non-zero. I am still trying to produce it with EKF in SITL. If i can, i will inform you again. Btw, i use EKF at hardware tests as seen in logs.