PX4/PX4-Autopilot #11503: [Err] [gazebo_mavlink_interface.cpp:1018] poll timeout when mc_att_control stop

srmx, 2019-02-19:
**Describe the bug**
I launch the simulation by
`make px4_sitl gazebo`
and right after that i am trying to stop attitude controller by typing 
`mc_att_control stop`
This leads to the repeating errors
```
[Err] [gazebo_mavlink_interface.cpp:1018] poll timeout
[Err] [gazebo_mavlink_interface.cpp:1018] poll timeout
...
```

**To Reproduce**
1. `make px4_sitl gazebo`
2. `mc_att_control stop`

**Expected behavior**
No errors

**Additional context**
I am using the px4 of master branch commit 18f7ee2d50dc45044008fc7ed323dabc46833236
Gazebo10
Ubuntu 18

a6a3uh, 2019-04-22:
The issue is still there with latest master branch c50c44cc582956aefd32e39a6c7a5210ba4e1585.
Why is it closed?

a6a3uh, 2019-04-22:
Btw. This is something introduced with v1.9. In v1.8.0-beta2  there is no such bug.

julianoes, 2019-04-23:
If you stop the attitude controller, no more attitude setpoints are published and therefore the simulation (e.g. Gazebo) times out because it is waiting for the latest motor commands. This happens because the SITL simulation now works in lockstep, so px4 and the simulation wait on the completion of each other.

To disable lockstep for your use case, set this:
https://github.com/PX4/Firmware/blob/c50c44cc582956aefd32e39a6c7a5210ba4e1585/boards/px4/sitl/default.cmake#L104

I'm assuming you want to use your own attitude controller. In that case I suggest that you change the startup script to never start `mc_att_control` and start your own controller instead. That would be hear:
https://github.com/PX4/Firmware/blob/c50c44cc582956aefd32e39a6c7a5210ba4e1585/ROMFS/px4fmu_common/init.d/rc.mc_apps#L55

a6a3uh, 2019-04-23:
I tried to do as you suggest. But stopping `mc_att_control` now gives a different kind of error:
```
ERROR [sensors] Accel #1 fail:  STALE!
ERROR [sensors] Sensor Accel #1 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 1: Accel #0 priority: 75
WARN  [sensors] Remaining sensors after failover event 1: Accel #1 priority: 1
```

If I not start `mc_att_control` at all as you suggested all is fine. But if it started and I stopping it the errors are as above.

This is worrying me as I tried a different feature. I've sent `mavlink_msg_set_actuator_control_target_pack` and get all the same error messages as above as well as Gyro and Mag and Baro sensors complains.

julianoes, 2019-04-23:
Ok, so I think if you want to use offboard where you send the actuator controls directly. This means that you either need to send the actuator controls in lockstep with receiving motor commands from PX4, or you need to disable lockstep as I suggested.

a6a3uh, 2019-04-23:
All described above is with lockstep disabled. What changed is I getting `SenorX fail: STALE!` instead of `poll timeout` now.

julianoes, 2019-04-23:
Ok not sure then. I'll try to reproduce.

julianoes, 2019-04-24:
Ok, I could reproduce it. I forgot that you also need to turn off lockstep in the Gazebo plugin.

For iris here: https://github.com/PX4/sitl_gazebo/blob/1dfa1652caaa08722a41047a9822889c73e42abb/models/rotors_description/urdf/iris_base.xacro#L20

For other models (e.g. plane) here: https://github.com/PX4/sitl_gazebo/blob/1dfa1652caaa08722a41047a9822889c73e42abb/models/plane/plane.sdf#L447

a6a3uh, 2019-04-24:
Now it works.) Thanks a lot!

closed 2019-04-24 by julianoes

wickedticket, 2022-12-11:
Hey @julianoes @msurov ! 

I'm new to this and have been facing a similar problem while trying to run my own application in JMAVSim. 

How did I get to this point?
1. I followed the Writing your first application tutorial from [here](https://dev.px4.io/v1.11_noredirect/en/apps/hello_sky.html)
2. I found an example [here](https://github.com/PX4/PX4-Autopilot/blob/main/src/examples/hwtest/hwtest.c) and modified it.

With the help of this thread here, I've done the following-
1. I've disabled lockstep in sitl.cmake in boards/px4/sitl
2. I've commented the line mc_rate_control start in PX4-Autopilot/ROMFS/px4fmu_common/init.d to disable the default application that controls the quad. 

Moreover, this is the code that I'm running in px4_simple_app to control the quad:

    struct actuator_controls_s actuators;
    memset(&actuators, 0, sizeof(actuators));
    orb_advert_t actuator_pub_ptr = orb_advertise(ORB_ID(actuator_controls_0), &actuators);

    struct actuator_armed_s arm;
    memset(&arm, 0, sizeof(arm));

    arm.timestamp = hrt_absolute_time();
    arm.ready_to_arm = true;
    arm.armed = true;
    orb_advert_t arm_pub_ptr = orb_advertise(ORB_ID(actuator_armed), &arm);
    orb_publish(ORB_ID(actuator_armed), arm_pub_ptr, &arm);

    /* read back values to validate */
    int arm_sub_fd = orb_subscribe(ORB_ID(actuator_armed));
    orb_copy(ORB_ID(actuator_armed), arm_sub_fd, &arm);

    if (arm.ready_to_arm && arm.armed) {
        warnx("Actuator armed");

    } else {
        errx(1, "Arming actuators failed");
    }

    hrt_abstime stime;

    int count = 0;

    while (count != 36) {
        stime = hrt_absolute_time();

        while (hrt_absolute_time() - stime < 1000000) {
            actuators.control[0]=0.5f;
            actuators.control[1]=0.3f;
            actuators.control[2]=0.3f;
            actuators.control[3]=0.9f;

            actuators.timestamp = hrt_absolute_time();
            orb_publish(ORB_ID(actuator_controls_0), actuator_pub_ptr, &actuators);
            usleep(10000);
        }

        warnx("count %i", count);
        count++;

There are no errors, but I don't see the quad flying or change yaw/pitch angles. What should I do? Am i missing something basic?