PX4/PX4-Autopilot #11507: GPS Issues_ 4 mini v5_ Pixhawk Pro v4 pro

Tony3dr, 2019-02-19:
When setting to Mission the vehicle behaved strangely, it was not following the waypoints(moved from left to right(disoriented). We tried to switch to position and it rejected the mode, also tried to RTL and it did not respond to the command. 

Steps to reproduce the behavior:
1. Drone switched on 
2. Uploaded mission 
3. Took off(smooth takeoff)
4. The vehicle did not follow the waypoint(moved left to right)
5. GPS was lost
7. Switched to position and it rejected the mode, tried to RTL and it did not respond to the command

The vehicle should have followed the waypoints 

Pixhawk 4 mini v5, **during flight the GPS failed** (Cloudy weather)
https://review.px4.io/plot_app?log=bd60a00f-6f68-4174-b793-268c5b8d204a

Tested on Pixhawk Pro v4pro, **during the mission GPS was lost and it did not follow the waypoints (Moved from left to right, disoriented (CLOUDY AND WINDY WEATHER).**
https://review.px4.io/plot_app?log=6a0d4bb5-1f85-4282-b6ed-5f68c39b0f31

julianoes, 2019-07-10:
Sorry for the late response. Have you found the reason for the GPS problems in the meantime? What I find interesting is that the GPS module reports fix type  = 4 which would mean RTK (floating) instead of only 3 (3D fix).

Tony3dr, 2019-07-15:
> Sorry for the late response. Have you found the reason for the GPS problems in the meantime? What I find interesting is that the GPS module reports fix type = 4 which would mean RTK (floating) instead of only 3 (3D fix).

@julianoes We haven't seen the issue we were unable to reproduce the issue. We will keep an eye out tho. Thanks.

closed 2019-07-18 by julianoes