PX4/PX4-Autopilot #11514: Update MAVLink RSSI field interpretation to latest versions

hamishwillee, 2019-02-21:
The  MAVLink definition of `.rssi` field on the following messages has been redefined from `0-100, 255=not used` to `0-254, 255=not used`: [RC_CHANNELS](https://mavlink.io/en/messages/common.html#RC_CHANNELS), [RC_CHANNELS_RAW](https://mavlink.io/en/messages/common.html#RC_CHANNELS_RAW), [RC_CHANNELS_SCALED](https://mavlink.io/en/messages/common.html#RC_CHANNELS_SCALED), [RADIO_STATUS](https://mavlink.io/en/messages/common.html#RADIO_STATUS)

Action is to move to the new definition. @LorenzMeier who best to assign to this?

Reference: https://github.com/mavlink/mavlink/issues/1027

hamishwillee, 2019-06-27:
This needs to be done.

julianoes, 2019-09-25:
By the way. I hat the bot.

julianoes, 2026-08-31:
This one is for you @hamishwillee