PX4/PX4-Autopilot #11522: MC motors fail to spin after transition back for my VTOL tailsitter

mrkhan200, 2019-02-22:
I have a VTOL tailsitter (with a + MC config). Out 0 & 1 are FW motors and Out 2 & 3 are MC motors and Out 4 & 5 are FW elevons. I am running an old PX4 firmware (2016) on Pixhawk 1 (from mRo). 

In front transition, I close the MC motors. In transition back, I turn ON the MC motors again. Unfortunately, it has happened to me twice that the MC motors did not spin back up again. I suspect this because even though I can see the PH sending PWM commands, the UAV enters into a pitch tumbling and crashes. Also, the props on the MC motors are intact after the crash indicating that they were not spinning at the time of crash.

I was wondering is there a possibility that while the pixhawk is logging PWM values, some other module might be calling and over-riding the PWM values without them being logged. Any insight into the fmu and pwm drivers will be appreciated.

Thanks.

[image]

julianoes, 2019-07-15:
@mrkhan200 sorry for the late response. Could you provide a link to a log please? Otherwise it's hard to answer. Also did you do any handheld tests without props to see if you can reproduce it?

julianoes, 2019-10-14:
Closing because there is no log.

closed 2019-10-14 by julianoes