PX4/PX4-Autopilot #11523: CI: Xvfb not starting

lamping7, 2019-02-22:
This might actually be better suited for an issue in PX4/containers, but here it is.

On my local Jenkins machine I can check the output from running Xvfb in the Docker entrypoint script. Here's what I see:
> _XSERVTransmkdir: ERROR: euid != 0,directory /tmp/.X11-unix will not be created.

I'm thinking we have a permission problem. The docker plugin for Jenkins automatically runs the container as the current user. Currently in CI this is uid:gid 1000:1000. On my machine it is different, but it doesn't matter. I think the same thing is happening in CI. This user isn't root. So, Xvfb can't start.

Additionally, adding the 1001 jenkins user in the docker file doesn't seem proper knowing this now.

@dagar if you can get into a running container and execute `docker logs` you can see the output from the Xvfb command.

Antiheavy, 2019-02-22:
I'm seeing a similar problem (errors) when running trying to set up SITL in a Docker for Windows.  I don't think I'm using Xvfb in my application anyway, but something isn't happy with how I was installing and running things in the container.

lamping7, 2019-02-22:
What is your error?

Antiheavy, 2019-02-22:
> What is your error?

[image]

lamping7, 2019-02-23:
@Antiheavy That has been fixed with https://github.com/PX4/containers/commit/31dc86bd112d3aa28ad44c6dee7f0cbd93c83728. We need to update the container tags for it to be realized on your end.

lamping7, 2019-03-07:
After some digging I found two problems. I'm just going to note them here before I work on an actual fix.

1. Permission error for xvfb -- this can be fixed pretty easily
2. DISPLAY env var isn't persisted when set in the `entrypoint` script -- This requires more thought on the best way to do this. We need another way of setting it and therefore a different way to trigger Xfvb to run in the `entrypoint`.

After some hackery, I got it to work as it should, but the method I used is not good for mainstream. More to come.

lamping7, 2019-03-07:
There are two ways to approach this. Either use it all the time (Fix in dockerfile) or only want to used in Jenkins (fix in Jenkinsfile and entrypoint). Any thoughts on this?

mrivi, 2019-03-07:
@lamping7  thanks for looking into this. 
FYI @dagar

lamping7, 2019-03-07:
See https://github.com/PX4/containers/pull/179. It was decided to use Xvfb by default, but allow users to override the `DISPLAY` with `docker run -it -e DISPLAY=${DISPLAY} ...` as show in the existing [README](https://github.com/PX4/containers/blob/master/docker/px4-dev/README.md). This means Jenkins will be able to use it without any changes over here in Firmware.

closed 2019-03-07 by dagar