PX4/PX4-Autopilot #11530: Copter cant fly with following z setpoint

E666GT, 2019-02-23:
I make some changes in Firmware/src/ pos control module. I set up a setpoint of z, then I thought I maybe see the copter fly with the setpoint. For example, I set z＝-3, but it obviously flies much much higher. What's the supposed problem？

Hardware Pixhawk
Version: 1.9.0a

E666GT, 2019-02-23:
Here is the DIY code, start from line 206th.
https://github.com/E666GT/Firmware/blob/v190a_setout/src/modules/mc_pos_control/PositionControl.cpp

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-15:
Sorry for the late response here but I'm not sure anyone has time to review or debug your custom code here. Hopefully you could resolve your issue in the meantime.

If you have this issue with the default/unchanged code than that would be something we would have to have a closer look at.