PX4/PX4-Autopilot #11544: Wrong UDP Port when using -t argument

hxdnshx, 2019-02-26:
**Describe the bug**
When I use `-t IPAddress` argument of command `mavlink start`, pixhawk(SITL) connect to GCS automatically, but with a random port number(such as 49224), rather than the value of argument `-u PortNumber` or the default value. This unwanted behaviour cause GCS unable to send any command to pixhawk.

**To Reproduce**
Steps to reproduce the behavior:
1. Luanch a px4 docker container
>docker run -it --privileged -p 14562:14556/udp --name=mycontainer px4io/px4-dev-ros:lat
est bash

2. Clone px4 firmware
>git clone --recursive https://github.com/PX4/Firmware.git

3. Build,Source and launch(From [documention](https://dev.px4.io/v1.8.2/en/simulation/multi-vehicle-simulation.html))
>cd Firmware/
>make posix_sitl_default gazebo

4. Start a new mavlink connection in px4 shell
>px4> mavlink start -x -u 14557 -t 192.168.1.2 -o 14550 -r 4000000

**Expected behavior**
Pixhawk connect to GCS with correct port.

bkueng, 2019-02-28:
> Note:It seems only occurs in Windows Docker?

This could indeed be docker-related. The code does not do anything different in terms of ports when you specify a target IP. Can you look at the network traffic?

hxdnshx, 2019-03-01:
You're right.
I set the wrong host IP which isn't in the Docker network. So every udp packet send to host are processed by NAT.
In conclusion, for any users who using `-t` parameter, you should use the host IP Address in 'vEthernet (DockerNAT)' network.
Thanks.

closed 2019-03-01 by hxdnshx