PX4/PX4-Autopilot #11552: Error make aerofc-v1_default on px4_atomic.h

NLag, 2019-02-27:
**Describe the bug**
I can't "make intel_aerofc-v1_default" on master (tag 1.9.0-beta). There is an error on px4_atomic.h.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the Firmware with --recursive
2. "make intel_aerofc-v1_default"
3. See error

**Log Files and Screenshots**
**Output**
[1/388] Building CXX object src/module...s/modules__dataman.dir/dataman.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DCONFIG_ARCH_BOARD_INTEL_AEROFC_V1 -DMODULE_NAME=\"dataman\" -DPX4_MAIN=dataman_app_main -D__CUSTOM_FILE_IO__ -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I../../boards/intel/aerofc-v1/src -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../src/platforms/common -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -funsafe-math-optimizations -fno-strict-aliasing -fno-math-errno -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -D__STDC_VERSION__=199901L -nostdinc++ -Wframe-larger-than=1200 -std=gnu++11 -MMD -MT src/modules/dataman/CMakeFiles/modules__dataman.dir/dataman.cpp.obj -MF src/modules/dataman/CMakeFiles/modules__dataman.dir/dataman.cpp.obj.d -o src/modules/dataman/CMakeFiles/modules__dataman.dir/dataman.cpp.obj -c ../../src/modules/dataman/dataman.cpp
In file included from ../../src/platforms/px4_module.h:44:0,
                 from ../../src/modules/dataman/dataman.cpp:46:
../../src/platforms/px4_atomic.h: In instantiation of 'class px4::atomic<bool>':
../../src/platforms/px4_module.h:421:42:   required from here
../../src/platforms/px4_atomic.h:72:2: error: non-constant condition for static assertion
  static_assert(__atomic_always_lock_free(sizeof(T),(char*) 0), "atomic is not lock-free for the given type T");
  ^
compilation terminated due to -Wfatal-errors.
ninja: build stopped: subcommand failed.
Makefile:190: recipe for target 'intel_aerofc-v1_default' failed
make: *** [intel_aerofc-v1_default] Error 1

**Additional context**
I found that there is a patch [C/C++ PATCH] Allow __atomic_always_lock_free in a static assert (PR c/62024). It suppose to be ok.
I wonder if the error comes because of my gcc version.
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
arm-none-eabi-g++ (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.

NLag, 2019-02-27:
I fixed the issue. The problem is because of the version of gcc-arm-none-eabi. 
I install it with sudo apt-get (current 4.9)
I download a newer version (7.0) from 
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads 
Extract, add to $PATH, make a symlink in /usr/bin/arm-none-eabi-g++
And "make intel_aerofc-v1_default" again to fix the problem.

closed 2019-02-27 by NLag

aogrcs, 2019-03-01:
Hi, NLag
It did work, thanks

swrdZWJ, 2019-03-29:
Hello,I have the same problem with you
 And I download a newer version (7.0) 
then I add export PATH= $PATH:"/home/zhu/gcc-arm-none-eabi-7-2018-q2-update/bin"   in my bashrc
but I cannot solve the problem

error is:   /bin/sh: /usr/bin/arm-none-eabi-gcc: no such file or directory

NLag, 2019-04-01:
@swrdZWJ in your case, you can try make symlink from /usr/bin/arm-none-eabi-gcc to the new verison /bin/arm-none-eabi-gcc folder. Make symlink of one files instead of symlink all the folder.
You can try 2 ways:
sudo ln -s /home/zhu/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc /usr/bin/arm-none-eabi-gcc
sudo ln -s /home/zhu/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-g++ /usr/bin/arm-none-eabi-g++

swrdZWJ, 2019-04-01:
HI, NLag

It works well, thank you very much

sandeshneupane, 2021-11-15:
Hi,
I followed all the suggested steps above. 

Still i am getting these errors--

[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250_AK8963.cpp.o
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
                 from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.hpp:46,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic<long long unsigned int>’:
/home/pi/PX4-Autopilot/src/drivers/imu/invensense/icm20948/ICM20948.hpp:172:51:   required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
  static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/build.make:63: src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/ICM20948.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:6616: src/drivers/imu/invensense/icm20948/CMakeFiles/drivers__imu__invensense__icm20948.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
                 from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.hpp:46,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic<long long unsigned int>’:
/home/pi/PX4-Autopilot/src/drivers/imu/invensense/mpu9250/MPU9250_I2C.hpp:139:51:   required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
  static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/build.make:63: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/MPU9250_I2C.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic_bitset.h:36,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceMaster.hpp:5,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/uORBDeviceNode.hpp:37,
                 from /home/pi/PX4-Autopilot/platforms/common/uORB/PublicationMulti.hpp:4,
                 from /home/pi/PX4-Autopilot/src/lib/drivers/accelerometer/PX4Accelerometer.hpp:39,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.hpp:47,
                 from /home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.cpp:34:
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘class px4::atomic<long long unsigned int>’:
/home/pi/PX4-Autopilot/src/drivers/imu/analog_devices/adis16448/ADIS16448.hpp:115:51:   required from here
/home/pi/PX4-Autopilot/platforms/common/include/px4_platform_common/atomic.h:76:41: error: static assertion failed: atomic is not lock-free for the given type T
  static_assert(__atomic_always_lock_free(sizeof(T), 0), "atomic is not lock-free for the given type T");
                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/build.make:63: src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/ADIS16448.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250_i2c.dir/mpu9250_i2c_main.cpp.o
[ 96%] Building CXX object src/drivers/imu/analog_devices/adis16448/CMakeFiles/drivers__imu__analog_devices__adis16448.dir/adis16448_main.cpp.o
[ 96%] Building CXX object src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250.cpp.o
[... cut ...]
                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[3]: *** [src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/build.make:76: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/MPU9250.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:6724: src/drivers/imu/invensense/mpu9250/CMakeFiles/drivers__imu__invensense__mpu9250.dir/all] Error 2
make[1]: *** [Makefile:152: all] Error 2
make: *** [Makefile:230: emlid_navio2] Error 2

NLag, 2021-11-15:
Hello, my issue was from 2019. It's been 2 years, I don't work on this project any more and I suppose there have been many changes. But, I have a few suggestions for you:
- as I can see, you rename the /usr/bin/arm-none-eabi-gcc and g++, which doesn't have the same effect as symlink. You can try to make the symlink work or figure out what prevent the symlink. 
- you can try using version 7.0 as in my issue. you can check if there exists multiple versions of arm-none-aebi-gcc that might create the problem. 
- your issue relate to different make setting than mine, but if possible, you can look for exact compile line. In my case:
 FAILED: /usr/bin/arm-none-eabi-g++ -DCONFIG_ARCH_BOARD_INTEL_AEROFC_V1 -DMODULE_NAME="dataman" -DPX4_MAIN=dataman_app_main -D__CUSTOM_FILE_IO__ -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem

sandeshneupane, 2021-11-17:
Thanks! @NLag

johnBrereton, 2023-09-08:
@sandeshneupane I am running into the same issue as you, I know its been a while but do you recall if you were able to find a fix?  It would be a huge help.

VmSupport, 2024-01-21:
Hi, this thread is quite old already but then starting a new one wouldn't make any sense I guess..

This issue is not common known I think, not sure if there are a lot of developers working with RPi 4/CM4 (PilotPi).
I'm having the exact same issue as @johnBrereton and @sandeshneupane with the same output he posted way earlier.
Maybe @johnBrereton has found something or hopefully someone like @NLag can point us in the right direction.

gcc (Raspbian 12.2.0-14+rpi1) 12.2.0
g++ (Raspbian 12.2.0-14+rpi1) 12.2.0

Both arm-none-eabi-gcc and g++ exist in /bin and /usr/bin

johnBrereton, 2024-01-21:
@VmSupport Unfortunately, I was not able to find a fix for this issue. Between this and several other issues I encountered trying to get PX4 running on a Pi, I decided to switch to arduPilot and have never looked back. It seems like the Pi is not a priority for the PX4 team and many of the support documents and packages are very out of date. Since switching to ArduPilot, my Pi4 has worked flawlessly and I am still able to use Q Ground Control.

NLag, 2024-01-22:
@VmSupport 

I'm not sure my knowledge about this project is still suitable to help you but I have a few suggestions.
You might need to check the exact error. Mine was "error: non-constant condition for static assertion" which meant __atomic_always_lock_free check was not usable in static assertion. This was solved in one of C/C++ patch so updating gcc/g++ fixed it. 

@sandeshneupane's error was "error: static assertion failed: atomic is not lock-free for the given type T" which originally different from mine.The issue might be caused by `make` target :
Different `make` target build for different machine. Mine was a default build for a simulation on a ubuntu desktop machine. Other build may have this problem (e.g ARM for Raspberry Pi has reported lock-free inconsistancy)
https://stackoverflow.com/questions/64242067/stdatomicbool-lock-free-inconsistency-on-arm-raspberry-pi-3 

The C++ standard allows for types to be "sometimes lock free": maybe it is not known at compile time whether std::atomic<T> will be lock-free, because it depends on machine features that will be detected at runtime.

Changes in new atomic.h lib may break old build target. (mismatch version problem).

VmSupport, 2024-01-24:
@NLag thanks for your reply.. I had the same thoughts about gcc updating and build versions, version 12 is the latest.
Found a similar update issue with empy (pip install) which is now at version 4.0 and had to downgrade to version empy==3.3.4

@johnBrereton finally I got a working build process with Ubuntu armhf 20.04.6 LTS on a RPi 4 and CM4 and PX4 running.