PX4/PX4-Autopilot #11568: Error while executing "make px4_sitl gazebo"

canberkgurel, 2019-03-01:
I followed the setup instructions on http://dev.px4.io 

When I run 
```

cd ~/src/Firmware
make px4_sitl jmavsim
```

it all works fine but when I do the same with gazebo,

```
cd ~/src/Firmware
make px4_sitl gazebo
```

I am receiving the following error:

```
Makefile:197: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```
I am using Ubuntu 16.04 (and ROS Kinect). The full output is as follows: 
```

[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.14
-- Using C++14 compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo
[3/6] Performing build step for 'sitl_gazebo'
FAILED: cd /home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/csgurel/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/41] Building CXX object CMakeFiles/sensor_msgs.dir/Imu.pb.cc.o
[2/41] Building CXX object CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o
[3/41] Linking CXX shared library libmav_msgs.so
[4/41] Linking CXX shared library libstd_msgs.so
[5/41] Building CXX object CMakeFiles/physics_msgs.dir/Wind.pb.cc.o
FAILED: : && /usr/bin/c++  -fPIC -Wno-deprecated-declarations   -shared -Wl,-soname,libmav_msgs.so -o libmav_msgs.so CMakeFiles/mav_msgs.dir/CommandMotorSpeed.pb.cc.o CMakeFiles/mav_msgs.dir/MotorSpeed.pb.cc.o -L/usr/lib/x86_64-linux-gnu/gazebo-7/plugins  -L/home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_timer.so /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so /usr/lib/x86_64-linux-gnu/libgazebo_physics.so /usr/lib/x86_64-linux-gnu/libgazebo_ode.so /usr/lib/x86_64-linux-gnu/libgazebo_transport.so /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so /usr/lib/x86_64-linux-gnu/libgazebo_util.so /usr/lib/x86_64-linux-gnu/libgazebo_common.so /usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so /usr/lib/x86_64-linux-gnu/libgazebo_opcode.so /usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.so /usr/lib/x86_64-linux-gnu/libgazebo_math.so /usr/lib/x86_64-linux-gnu/libgazebo_ccd.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_timer.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/local/lib/libprotobuf.a -lpthread /usr/lib/x86_64-linux-gnu/libsdformat.so /usr/lib/x86_64-linux-gnu/libignition-math2.so -lOgreMain -lpthread /usr/lib/x86_64-linux-gnu/libignition-math2.so /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_stitching3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_superres3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_videostab3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_aruco3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_bgsegm3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_bioinspired3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ccalib3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_cvv3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dpm3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_face3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_fuzzy3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_hdf3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_img_hash3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_line_descriptor3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_optflow3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_reg3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_rgbd3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_saliency3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_stereo3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_structured_light3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_surface_matching3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_tracking3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xfeatures2d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ximgproc3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xobjdetect3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xphoto3.so.3.3.1 /usr/local/lib/libprotobuf.a /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/libgazebo_rendering.so /usr/lib/x86_64-linux-gnu/libgazebo_physics.so /usr/lib/x86_64-linux-gnu/libgazebo_ode.so /usr/lib/x86_64-linux-gnu/libgazebo_transport.so /usr/lib/x86_64-linux-gnu/libgazebo_msgs.so /usr/lib/x86_64-linux-gnu/libgazebo_util.so /usr/lib/x86_64-linux-gnu/libgazebo_common.so /usr/lib/x86_64-linux-gnu/libgazebo_gimpact.so /usr/lib/x86_64-linux-gnu/libgazebo_opcode.so /usr/lib/x86_64-linux-gnu/libgazebo_opende_ou.so /usr/lib/x86_64-linux-gnu/libgazebo_math.so /usr/lib/x86_64-linux-gnu/libgazebo_ccd.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/local/lib/libprotobuf.a -lpthread /usr/lib/x86_64-linux-gnu/libsdformat.so /usr/lib/x86_64-linux-gnu/libignition-math2.so -lOgreMain -lpthread /usr/local/lib/libprotobuf.a -lpthread /usr/lib/x86_64-linux-gnu/libsdformat.so /usr/lib/x86_64-linux-gnu/libignition-math2.so -lOgreMain -lpthread /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_shape3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_photo3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_datasets3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_plot3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_text3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ml3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_video3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_calib3d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_features2d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_highgui3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_videoio3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_viz3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_phase_unwrapping3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_flann3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_objdetect3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgproc3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3.1 -Wl,-rpath,/usr/lib/x86_64-linux-gnu/gazebo-7/plugins:/home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo:/opt/ros/kinetic/lib/x86_64-linux-gnu: && :
/usr/bin/ld: /usr/local/lib/libprotobuf.a(common.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libprotobuf.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
FAILED: : && /usr/bin/c++  -fPIC -Wno-deprecated-declarations   -shared -Wl,-soname,libstd_msgs.so -o libstd_msgs.so CMakeFiles/std_msgs.dir/Int32.pb.cc.o -L/usr/lib/x86_64-linux-gnu/gazebo-7/plugins  -L/home/csgurel/src/Firmware/build/px4_sitl_default/build_gazebo /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_timer.so /usr/lib/x86_64-linux-gnu/libgazebo.so /usr/lib/x86_64-linux-gnu/libgazebo_client.so /usr/lib/x86_64-linux-gnu/libgazebo_gui.so /usr/lib/x86_64-linux-gnu/libgazebo_sensors.so /usr/lib/x86_64-linux-gnu/lib [... cut ...]
Odometry.pb.cc:127:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
Odometry.pb.cc:129:7: error: ‘internal_default_instance’ is not a member of ‘gazebo::msgs::Vector3d’
       ::gazebo::msgs::Vector3d::internal_default_instance());
       ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:197: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

```

What am I missing? Thanks!

TSC21, 2019-03-01:
You have a collision between the protobuf libraries. I recommend you to uninstall the Gazebo version you install from deb packages, as also the protobuf and libprotobuf. And just use the ones that come with ROS.

closed 2019-06-24 by canberkgurel

bigbellmercy, 2020-09-10:
> You have a collision between the protobuf libraries. I recommend you to uninstall the Gazebo version you install from deb packages, as also the protobuf and libprotobuf. And just use the ones that come with ROS.

The above solution succeeded to me. The ROS/Gazebo installation procedure is [here](https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#rosgazebo).

In the installation, in the `ubuntu_sim_ros_melodic.sh` script, I changed `catkin build` to `catkin_make_isolated` and `~/catkin_ws/devel` to `~/catkin_ws/devel_isolated` because some already existing ROS packages cannot be built by `catkin build` command.