PX4/PX4-Autopilot #11569: Pixhawk 4 and Pikhawk 4mini v5 not disarming.

Junkim3DR, 2019-03-01:
**Description**
UAV is not disarming manually nor automatically, same issue when using Kill Switch.
Changed to Stable firmware and disarm functions where working as expected.

**Log Files**
Pixhawk 4mini v5
https://review.px4.io/plot_app?log=b69fc363-d90d-49d1-8cf4-ecb6122f3202&fbclid=IwAR2DQunVucBs96HvJiQ9KvGzdutroVqC33kkPLZk8crRz0QyFi97vb1qy2Q

Pikhawk 4 v5
https://review.px4.io/plot_app?log=e715787a-7f9f-45d4-9947-e5b0e1576773&fbclid=IwAR3dsmfYIADvrCuK_24NColLyb5a2E1UnXGX25eYMjg21_FRJSy3evWkG_4

**Drone**
- Multicopter.
- DJI Flame Wheel F450.

dagar, 2019-03-01:
How were you attempting to disarm? Do you recall the last time this was working correctly (without going all the way back to stable)?

dagar, 2019-03-01:
The kill switch was bouncing. What's the history of these exact setups?

[image]

Other oddities.
 - the estimator stops responding on the first attempted disarm. 
[image]
 - no cpuload

dagar, 2019-03-01:
Just noticed the logger watchdog activated (which explains some of the other problems), but doesn't tell us what the core problem is.

[image]

dagar, 2019-03-02:
There's something running in LPWORK that's misbehaving.

[image]

dagar, 2019-03-02:
@Junkim3DR could you comment on the status lights and buzzer? Can you verify if they're working or not? Any differences compared to stable?

dagar, 2019-03-02:
It's something related to `mtd` on fmu-v5. The param save is blocking (LPWORK).

dagar, 2019-03-02:
Last working commit. https://github.com/PX4/Firmware/commit/0d170a0eaf51adec6deb550642228f80c97b3c1f

dagar, 2019-03-02:
@davids5 https://github.com/PX4/Firmware/commit/0846059646990fd4fbe4a884dbd1a8ff46d559c1 is the culprit. 

Many thanks to @mcsauder for the hands on debugging.

dagar, 2019-03-02:
GPIO_nARMED conflicts with SPI2 SCK (FRAM).

[image]

[image]

mcsauder, 2019-03-02:
@dagar , #11575.

Thanks for localizing that!

dagar, 2019-03-02:
Fixed by https://github.com/PX4/Firmware/pull/11575

closed 2019-03-02 by dagar