PX4/PX4-Autopilot #11578: Cannot build code: make: *** [px4_fmu-v3_default] Error 1

guroz, 2019-03-03:
Good morning,

I am entirely new to PX4 and I am probably missing something but I have been trying for 2 days and still cannot find out how to address my problem. I have followed the developer guidelines to install everything and I am following the Getting Started tutorial. When I try to build the code with : 

`make px4_fmu-v2_default`

The code builds and everything is fine. However, I am using a Pixhawk 2.1 and therefore cannot upload this version to my board. When I try to build using 

`make px4_fmu-v3_default`

it returns this error : 

`[1140/1142] Linking CXX executable px4_fmu-v3_default.elf
FAILED: px4_fmu-v3_default.elf 
: && /usr/local/bin/arm-none-eabi-g++  -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -DNDEBUG  --specs=nosys.specs platforms/nuttx/CMakeFiles/px4.dir/__/__/src/platforms/empty.c.obj  -o px4_fmu-v3_default.elf -L/Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/Test/Firmware/build/px4_fmu-v3_default/external/Install/lib -nostartfiles -nodefaultlibs -nostdlib -nostdinc++ -fno-exceptions -fno-rtti -Wl,--script=/Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/Test/Firmware/build/px4_fmu-v3_default/NuttX/nuttx-config/scripts/ld.script -Wl,-Map=px4_fmu-v3_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group src/drivers/boards/common/stm32/libdrivers_boards_common_arch.a NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libs/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc src/lib/cdev/test/liblib__cdev__test__cdev_test.a src/lib/controllib/controllib_test/liblib__controllib__controllib_test.a src/modules/uORB/libmodules__uORB.a src/modules/uORB/uORB_tests/libmodules__uORB__uORB_tests.a src/drivers/barometer/bmp280/libdrivers__bmp280.a src/drivers/barometer/lps22hb/libdrivers__barometer__lps22hb.a src/drivers/barometer/ms5611/libdrivers__ms5611.a src/drivers/batt_smbus/libdrivers__batt_smbus.a src/drivers/camera_capture/libdrivers__camera_capture.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/differential_pressure/ets/libdrivers__ets_airspeed.a src/drivers/differential_pressure/ms4525/libdrivers__ms4525_airspeed.a src/drivers/differential_pressure/ms5525/libdrivers__ms5525_airspeed.a src/drivers/differential_pressure/sdp3x/libdrivers__sdp3x_airspeed.a src/drivers/distance_sensor/cm8jl65/libdrivers__cm8jl65.a src/drivers/distance_sensor/leddar_one/libdrivers__leddar_one.a src/drivers/distance_sensor/ll40ls/libdrivers__ll40ls.a src/drivers/distance_sensor/mb12xx/libdrivers__mb12xx.a src/drivers/distance_sensor/pga460/libdrivers__pga460.a src/drivers/distance_sensor/sf0x/libdrivers__sf0x.a src/drivers/distance_sensor/sf0x/sf0x_tests/libdrivers__sf0x__sf0x_tests.a src/drivers/distance_sensor/sf1xx/libdrivers__sf1xx.a src/drivers/distance_sensor/srf02/libdrivers__srf02.a src/drivers/distance_sensor/teraranger/libdrivers__teraranger.a src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a src/drivers/distance_sensor/ulanding/libdrivers__ulanding.a src/drivers/distance_sensor/vl53lxx/libdrivers__vl53lxx.a src/drivers/gps/libdrivers__gps.a src/drivers/imu/adis16448/libdrivers__adis16448.a src/drivers/imu/l3gd20/libdrivers__l3gd20.a src/drivers/imu/lsm303d/libdrivers__lsm303d.a src/drivers/imu/mpu6000/libdrivers__mpu6000.a src/drivers/imu/mpu9250/libdrivers__mpu9250.a src/drivers/irlock/libdrivers__irlock.a src/drivers/lights/blinkm/libdrivers__blinkm.a src/drivers/lights/oreoled/libdrivers__oreoled.a src/drivers/lights/rgbled/libdrivers__rgbled.a src/drivers/lights/rgbled_ncp5623c/libdrivers__rgbled_ncp5623c.a src/drivers/magnetometer/bmm150/libdrivers__bmm150.a src/drivers/magnetometer/hmc5883/libdrivers__hmc5883.a src/drivers/magnetometer/ist8310/libdrivers__ist8310.a src/drivers/magnetometer/lis3mdl/libdrivers__lis3mdl.a src/drivers/magnetometer/lsm303agr/libdrivers__magnetometer__lsm303agr.a src/drivers/magnetometer/rm3100/libdrivers__rm3100.a src/drivers/mkblctrl/libdrivers__mkblctrl.a src/drivers/lights/pca8574/libdrivers__pca8574.a src/drivers/pca9685/libdrivers__pca9685.a src/drivers/protocol_splitter/libdrivers__protocol_splitter.a src/drivers/pwm_input/libdrivers__pwm_input.a src/drivers/pwm_out_sim/libdrivers__pwm_out_sim.a src/drivers/px4flow/libdrivers__px4flow.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/px4io/libdrivers__px4io.a src/drivers/roboclaw/libdrivers__roboclaw.a src/drivers/stm32/adc/libdrivers__adc.a src/drivers/tap_esc/libdrivers__tap_esc.a src/drivers/telemetry/bst/libdrivers__bst.a src/drivers/telemetry/frsky_telemetry/libdrivers__frsky_telemetry.a src/drivers/telemetry/hott/hott_sensors/libdrivers__hott__hott_sensors.a src/drivers/telemetry/hott/hott_telemetry/libdrivers__hott__hott_telemetry.a src/drivers/telemetry/iridiumsbd/libdrivers__iridiumsbd.a src/drivers/test_ppm/libdrivers__test_ppm.a src/drivers/tone_alarm/libdrivers__tone_alarm.a src/drivers/uavcan/libmodules__uavcan.a src/modules/attitude_estimator_q/libmodules__attitude_estimator_q.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/commander/commander_tests/libmodules__commander__commander_tests.a src/modules/dataman/libmodules__dataman.a src/modules/ekf2/libmodules__ekf2.a src/modules/events/libmodules__events.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/gnd_att_control/libmodules__gnd_att_control.a src/modules/gnd_pos_control/libmodules__gnd_pos_control.a src/modules/land_detector/libmodules__land_detector.a src/modules/landing_target_estimator/libmodules__landing_target_estimator.a src/modules/load_mon/libmodules__load_mon.a src/modules/local_position_estimator/libmodules__local_position_estimator.a src/modules/logger/libmodules__logger.a src/modules/mavlink/libmodules__mavlink.a src/modules/mavlink/mavlink_tests/libmodules__mavlink__mavlink_tests.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/navigator/libmodules__navigator.a src/modules/sensors/libmodules__sensors.a src/modules/vmount/libdrivers__vmount.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/wind_estimator/libmodules__wind_estimator.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/config/libsystemcmds__config.a src/systemcmds/dumpfile/libsystemcmds__dumpfile.a src/systemcmds/esc_calib/libsystemcmds__esc_calib.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/led_control/libsystemcmds__led_control.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/motor_ramp/libsystemcmds__motor_ramp.a src/systemcmds/motor_test/libsystemcmds__motor_test.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/nshterm/libsystemcmds__nshterm.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/reflect/libsystemcmds__reflect.a src/systemcmds/sd_bench/libsystemcmds__sd_bench.a src/systemcmds/shutdown/libsystemcmds__shutdown.a src/systemcmds/tests/libsystemcmds__tests.a src/systemcmds/tests/hrt_test/libsystemcmds__tests__hrt_test.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/topic_listener/libsystemcmds__topic_listener.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/systemcmds/usb_connected/libsystemcmds__usb_connected.a src/systemcmds/ver/libsystemcmds__ver.a src/examples/bottle_drop/libmodules__bottle_drop.a src/examples/fixedwing_control/libexamples__fixedwing_control.a src/examples/hello/libexamples__hello.a src/examples/hwtest/libexamples__hwtest.a src/examples/position_estimator_inav/libmodules__position_estimator_inav.a src/examples/px4_mavlink_debug/libexamples__px4_mavlink_debug.a src/examples/px4_simple_app/libexamples__px4_simple_app.a src/examples/rover_steering_control/libexamples__rover_steering_control.a src/examples/segway/libexamples__segway.a src/examples/uuv_example_app/libexamples__uuv_example_app.a ROMFS/libromfs.a src/lib/drivers/smbus/libdrivers__smbus.a src/lib/drivers/airspeed/libdrivers__airspeed.a src/lib/drivers/accelerometer/libdrivers_accelerometer.a src/lib/drivers/gyroscope/libdrivers_gyroscope.a src/lib [... cut ...]
/usr/local/Cellar/gcc-arm-none-eabi/20180627/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find src/lib/perf/libperf.a: Too many open files
/usr/local/Cellar/gcc-arm-none-eabi/20180627/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find src/lib/parameters/tinybson/libtinybson.a: Too many open files
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v3_default] Error 1`

I am using macOS mojave 10.14.2 and just installed every toolchains, so they all are up to date.
Thank you in advance

LorenzMeier, 2019-03-03:
Here is how to fix this temporarily: http://discuss.px4.io/t/errors-about-too-many-open-files/9150/2

guroz, 2019-03-03:
Thank you very much, it worked !

closed 2019-03-03 by guroz