PX4/PX4-Autopilot #11585: Type of file airframe + error when building code UnicodeEncodeError: 'ascii' codec can't encode character '\xea' in position 38: ordinal not in range(128)

guroz, 2019-03-04:
Good evening,

I am new to px4 and I encountered an issue when trying to build a new airframe (following the tutorial [https://dev.px4.io/en/airframes/adding_a_new_frame.html](url)). The added airframe is : 

`#!nsh
#
# @name prao
#
# @type Flying Wing
# @class Plane
#
# @maintainer Johan Poccard <johan.poccard-saudart@epfl.ch>
#
# @output MAIN1 rudder
# @output MAIN2 main foil
# @output MAIN3 small foil
# @output MAIN4 throttle
#
# @output AUX A DEFINIR
#

sh /etc/init.d/rc.fw_defaults // peut-être besoin de créer un nouveau fichier defaults

if [$AUTOCNF = yes]
then
        param set FW_AIRSPD_MAX 10  //recopié du bifoiler
        param set FW_AIRSPD_MIN 3
        param set FW_AIRSPD_TRIM 5
        param set FW_ATT_TC 0.3
        param set FW_L1_DAMPING 0.74
        param set FW_L1_PERIOD 16
        param set FW_LND_ANG 15
        param set FW_LND_FLALT 5
        param set FW_LND_HHDIST 15
        param set FW_LND_HVIRT 13
        param set FW_LND_TLALT 5
        param set FW_THR_LND_MAX 0
        param set FW_PR_FF 0.35
        param set FW_PR_IMAX 0.4
        param set FW_PR_P 0.08
        param set FW_RR_FF 0.6
        param set FW_RR_IMAX 0.2
        param set FW_RR_P 0.04
        param set SYS_COMPANION 157600
        param set PWM_MAIN_REV0 1
        param set PWM_MAIN_REV1 1
        param set PWM_DISARMED 0
        param set PWM_MIN 1000
        param set PWM_MAX 2000
fi

//Configure this as fixed wing
set MAV_TYPE 1

//Set mixer
set MIXER prao

//PWM
set PWM_DISARMED p:PWM_DISARMED
set PWM_MIN p:PWM_MIN
set PWM_MAX p:PWM_MAX`

And the mixer associated is : 

`PRAO mixer
=================

Designed for EPFL PRAO

This file defines mixers suitable for controlling a prao boat suing PX4.
[... 330 lines cut ...]
  than in this calling directory.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware
[491/1141] Performing build step for 'px4io_firmware'
[0/198] git submodule platforms/nuttx/NuttX/nuttx
[1/198] git submodule platforms/nuttx/NuttX/apps
[198/198] Creating /Users/guillaumeroz...d/px4io_firmware/px4_io-v2_default.px4
[1133/1141] ROMFS: pruning
FAILED: ROMFS/romfs_pruned.stamp 
cd /Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/build/px4_fmu-v3_default/ROMFS && /anaconda3/bin/python /Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/Tools/px_romfs_pruner.py --folder /Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/build/px4_fmu-v3_default/genromfs --board px4_fmu-v3 && /usr/local/Cellar/cmake/3.13.4/bin/cmake -E touch romfs_pruned.stamp
Traceback (most recent call last):
  File "/Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/Tools/px_romfs_pruner.py", line 120, in <module>
    main()
  File "/Users/guillaumerozand/Documents/EPFL/MASTER2/ControlePRAO/GITHUB_PX4/Firmware/Tools/px_romfs_pruner.py", line 114, in main
    f.write(pruned_content.encode("ascii", errors='strict'))
UnicodeEncodeError: 'ascii' codec can't encode character '\xea' in position 38: ordinal not in range(128)
[1134/1141] Generating ../gencromfs
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v3_default] Error 1`
How can I make it work ?

Thank you in advance for your answer.

hamishwillee, 2019-03-04:
The error is exactly what it says - somewhere in your text you have a unicode character. The file format only supports ASCII. It is probably some accent character that isn't part of the standard set for English.

You will need to strip it out. What might work is using a tool like this: https://onlineasciitools.com/convert-unicode-to-ascii

guroz, 2019-03-05:
Thank you very much !

closed 2019-03-05 by guroz