PX4/PX4-Autopilot #11589: Unable to access to udp port

docHell, 2019-03-05:
I've created a virtual machine running Xubuntu 16.04.
I've correctly ( I suppose ) installed jmavsim and px4.
When  I digit "_sudo make posix_sitl_default jmavsim_" the simulator starts correctly and I can run some commands succesfully like "_commander takeoff_".

According to the instructions, to connect to the wifi I must run the simulator using a different command : 

_sudo make broadcast jmavsim_

Then I've to set manually a parmeter in the consolle the firtst time.

param set MAV_BROADCAST 1

The screen log is this :

>  ninja: Entering directory `/home/dronetest/Firmware/build/posix_sitl_broadcast'
> [0/1] cd /home/dronetest/Firmware/buil...st/Firmware/build/posix_sitl_broadcast
> SITL ARGS
> sitl_bin: /home/dronetest/Firmware/build/posix_sitl_broadcast/bin/px4
> rcS_path: posix-configs/SITL/init/ekf2
> debugger: none
> program: jmavsim
> model: none
> src_path: /home/dronetest/Firmware
> build_path: /home/dronetest/Firmware/build/posix_sitl_broadcast
> empty model, setting iris as default
> SITL COMMAND: /home/dronetest/Firmware/build/posix_sitl_broadcast/bin/px4 /home/dronetest/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/dronetest/Firmware/test_data
> INFO  [Unknown] Creating symlink /home/dronetest/Firmware/ROMFS/px4fmu_common -> /home/dronetest/Firmware/build/posix_sitl_broadcast/tmp/rootfs/etc
> 
> ______  __   __    ___ 
> | ___ \ \ \ / /   /   |
> | |_/ /  \ V /   / /| |
> |  __/   /   \  / /_| |
> | |     / /^\ \ \___  |
> \_|     \/   \/     |_/
> 
> px4 starting.
> 
> INFO  [Unknown] Calling startup script: bash etc/init.d-posix/rcS 0
> INFO  [dataman] Unknown restart, data manager file './dataman' size is 11405132 bytes
> INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
> Buildfile: /home/dronetest/Firmware/Tools/jMAVSim/build.xml
> 
> make_dirs:
> 
> compile:
> 
> create_run_jar:
> 
> copy_res:
> 
> BUILD SUCCESSFUL
> Total time: 0 seconds
> Options parsed, starting Sim.
> Starting GUI...
> 3D [dev] 1.6.0-pre12-daily-experimental daily
> 
> libEGL warning: DRI2: failed to authenticate
> Init MAVLink
> INFO  [simulator] Got initial simulation data, running sim..
> INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
> INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
> INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
> INFO  [logger] logger started (mode=all)
> INFO  [logger] Start file log
> INFO  [logger] Opened log file: ./log/2019-03-05/01_06_01.ulg
> INFO  [Unknown] Startup script returned successfully
> pxh> INFO  [mavlink] using network interface enp0s3, IP: 192.168.1.200
> INFO  [mavlink] with netmask: 255.255.255.0
> INFO  [mavlink] and broadcast IP: 192.168.1.255
> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
> INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
> INFO  [ecl/EKF] EKF commencing GPS fusion
> ERROR [Unknown] WARNING! TIME IS NEGATIVE! 158155213 vs 158155216
> 

 I've tried with some software ( in java or nodejs)  to connect, inside and outside the vm, but the " ECONNREFUSED" is always the result.

I've tried also to ping using nmap the port, but

for the 14556/7
> sudo nmap -p 14556 -sU 192.168.1.200
> 
> Starting Nmap 7.60 ( https://nmap.org ) at 2019-03-05 02:53 CET
> Nmap scan report for 192.168.1.200
> Host is up (0.00017s latency).
> 
> PORT      STATE         SERVICE
> 14556/udp open|filtered unknown
> MAC Address: 08:00:27:66:94:F7 (Oracle VirtualBox virtual NIC)
> 
> Nmap done: 1 IP address (1 host up) scanned in 0.93 seconds
 
instead for the 14550/40:

>  sudo nmap -p 14540 -sU 192.168.1.200
> 
> Starting Nmap 7.60 ( https://nmap.org ) at 2019-03-05 02:53 CET
> Nmap scan report for 192.168.1.200
> Host is up (0.00012s latency).
> 
> PORT      STATE  SERVICE
> 14540/udp closed unknown
> MAC Address: 08:00:27:66:94:F7 (Oracle VirtualBox virtual NIC)
> 
> Nmap done: 1 IP address (1 host up) scanned in 0.82 seconds

I'm trying now using "https://github.com/kvenux/nodegcs"

But the result is the same : ECONNREFUSED

Can you help me ?

docHell, 2019-03-05:
ERROR [Unknown] WARNING! TIME IS NEGATIVE! 140078086 vs 140078087
This is the problem

 
[image]

When the error appear, always after few minutes, I lost forever the connection ( It is impossible to reconnect )

How can I solve ?

docHell, 2019-03-25:
OK 
The only problem is that the simulator doesn't work well in a virtual machine.
I've solved this "libEGL warning: DRI2: failed to authenticate" enabling the 3d acceleration, but the port problem still remain..
But it works pretty well on phisical machine

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-15:
I don't understand why you are using `sudo` to start it in your setup, and I'm also confused what you're try to achieve with broadcast. Alternatively to using broadcast, you could add `-t IP` to the `mavlink start` command in the rcS file.