PX4/PX4-Autopilot #11591: mavlink forward question

ToppingXu, 2019-03-05:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2518

I think it should be "message_buffer_mark_read(read_count);"
@julianoes can you confirm?
Thanks

julianoes, 2019-03-12:
@ToppingXu can you add some context for me to understand please?

Here are some questions that you could try to answer:
What is the problem that you ran into? What was expected but what happened instead? Why  is it required? What is the change?

ToppingXu, 2019-03-12:
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2467

it will lose some data. "message_buffer_mark_read(available)"  this will mark all date readed but we only parsed one message.

julianoes, 2019-03-12:
Hmm.

Do you know what this does?
```
available = message_buffer_get_ptr((void **)&read_ptr, &is_part);
```

ToppingXu, 2019-03-12:
Correct me if l was wrong

ToppingXu, 2019-03-12:
But when parse the second part it will stop if got a full msg. And available value is the whole data len.

ToppingXu, 2019-03-13:
Sorry, l understand it now. Thanks for your support

julianoes, 2019-03-13:
So it's correct? I was going to have an in-depth look  when I have time.

julianoes, 2019-06-12:
Closing as stale.

closed 2019-06-12 by julianoes