PX4/PX4-Autopilot #11595: VotedSensorsUpdate::check_failover seems to misinterpret simultaneous failures

tstellanova, 2019-03-06:
What is the expected behavior of `VotedSensorsUpdate::check_failover` ? 
In testing today it appeared that simultaneous failure of two accels was interpreted as having one valid accel remaining:

```
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Accel #1 priority: 100
```

Are there existing unit tests that should cover this case?  It doesn't appear that `VotedSensorsUpdate` has unit tests.

julianoes, 2019-07-15:
@tstellanova this seems indeed wrong. Could you confirm that this is still an issue with latest master?

tstellanova, 2019-07-18:
It may be a few weeks before I can retest this, but it didn’t appear there is a unit test for this logic. IIRC the test case was: multiple redundant voted sensors on the same bus failed.

julianoes, 2019-07-19:
FYI @bkueng