PX4/PX4-Autopilot #11598: SITL, Model: plane, [Err] [gazebo_mavlink_interface.cpp:1166] Error opening serial device: open: No such file or directory

libi95, 2019-03-06:
**Describe the bug**
Sitl simulation of the plane model doesn't work properly, since the gazebo_mavlink_interface doesn't seem to be able to open the serial device.
 
http://discuss.px4.io/t/sitl-gazebo-plane-error-with-gazebo-mavlink-interface/9852

**To Reproduce**
Steps to reproduce the behavior:
execute: 
~/src/Firmware$ make -j4 px4_sitl_default gazebo_plane
on command line.

**Expected behavior**
usual Sitl simulation: Opening of Gazebo with world/model + conection to PX4 over QGC

**Drone (please complete the following information):**
- (Not relevant for issue but need SITL to alter model for) Project Dipper (focus project at ETH)
https://dipper.ethz.ch/

**Screenshots**

[image]
[image]
[image]
[image]
[image]

lamping7, 2019-03-07:
I'm not sure if this is your problem, but that error is usually caused when the PATHs aren't setup correctly. I'd need to test the make targets.

libi95, 2019-03-07:
After installing the latest version of Ubuntu (18.04 LTS) the Simulation worked properly. It then was an issue related to my system, so the original code must be working properly. Sry for that and thanks for your reply!!

closed 2019-03-08 by lamping7