PX4/PX4-Autopilot #11600: Problem sending command long

garfieldG, 2019-03-07:
Hello,

Trying to send command long from the pilot failed.

I've noticed that the problem is with "update_if_changed" method, which always returns falls, even when new "vehicle_command" orb is published.
But when I changed it to "update" method it solved the case.
I've also noticed that none of the other implementations in mavlink_messages uses "update_if_changed", but "update" instead.
https://github.com/PX4/Firmware/blob/d12cec81baaa8bf1650327f78aa542d73e0c5bdd/src/modules/mavlink/mavlink_messages.cpp#L489

Does anyone ever noticed it before? Is there a reason to use this particular method only in command long or is it a mistake? and why does the method always returns false? could there be a problem with the method?

garfieldG, 2019-03-12:
Any thoughts?

hamishwillee, 2019-03-19:
I think it very unlikely there is a problem with this method. [Under the hood](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_orb_subscription.cpp#L106) this would require that `orb_check()` is not detecting the update, and that is used all over the place.

There are also several uses of `update_if_changed()`. I don't _know_ why some use (always) `update()` and others use `update_if_changed()`. IN this chase I would guess it is because you're only ever sending a mavlink command from PX4 if you're forwarding it to another component - and we only want to do that when we have a new command.

Are you sure this send() function is actually being called?

garfieldG, 2019-03-24:
@hamishwillee
I'm sure it's being called.
As I said the update method did solved it and the command was sent (something that didn't happen with update_if_changed).
The only uses of update_if_changed() are :
-for command long in mavlink_meassages
-for vehicle command ack in mavlink_main
The use in update_if_changed is not that common and there might be a problem.

julianoes, 2019-07-15:
@garfieldG this method has recently changed. Could you test this again?

closed 2019-07-18 by julianoes