PX4/PX4-Autopilot #11602: Support roll, pitch, thrust and yaw rate in OFFBOARD

ajshank, 2019-03-08:
Several offboard attitude control implementations build on a `[roll, pitch, *yawrate*, thrust]` framework. It does not look like PX4 supports that (apologies if I'm wrong). Setting yawrate has no effect if an attitude quaternion is provided. At present, one has to set `IGNORE_ATTITUDE` for rates to take effect (which would prevent roll and pitch commands). Also, ignoring the rate on one axis ignores it on all axes [L1385, `handle_message_set_attitude_target()`](https://github.com/PX4/Firmware/blob/d12cec81baaa8bf1650327f78aa542d73e0c5bdd/src/modules/mavlink/mavlink_receiver.cpp#L1385).
In short, if you used "setpoint_raw/attitude" plugin from Mavros to send a packet with:
- attitude quaternion, thrust
- yaw-rate
- `IGNORE_ROLL_RATE, IGNORE_PITCH_RATE`

then yaw-rate is ignored as well. Tested with latest stable release on real hardware.

It would be useful to have PX4 choose yaw vs yaw-rate depending on the bit masks specified in Mavlink packets.

**A solution**
Check for the specific bit mask `set_attitude_target.type_mask & (1 << 2)` to see if yaw-rate was requested. If it is, then bypass the yaw field in the quaternion (have the attitude controller working on roll, pitch and yaw-rate). Similar to Arducopter's [parsing](https://github.com/ajshank/ardupilot/blob/9a85a7c9e2f02effc3cd02d12d529f2ea496ccf7/ArduCopter/GCS_Mavlink.cpp#L1106). This _might_ break an existing workflow that doesn't specifically ignore yaw-rate though. I'm not sure if there's a simple workaround to that.

caomxin, 2019-03-22:
I can control the simulated vehicle (px4 firmware) via dronekit by sending **set_attitude_target** message. The vehicle must in OFFBOARD mode and the msg has to be sent in a frequency faster than 2Hz.

ajshank, 2019-03-31:
@caomxin Not quite sure if it is "supposed to" maintain altitude when setting thrust to 0.5. That scaling is a function of the total mass, battery, motors, propellers, and a bunch of other things; and I'm not sure how much of a tuning you can do. You need an outer-loop that converts errors in altitude to (scaled) thrust values.
However, this is not related to this issue/feature-request -- which itself has, curiously, gotten no other response ..

closed 2019-07-13 by stale[bot]

julianoes, 2019-07-15:
I believe this has been fixed #10892.

ajshank, 2019-07-22:
@julianoes Briefly went over the commit - it appears so. Thanks! :)