PX4/PX4-Autopilot #11605: Log EKF output before from initialization

CarlOlsson, 2019-03-08:
The estimated states are not logged until EKF2 declares tilt_align = true. This is suboptimal since it makes it hard to discover issues with the EKF initialization such as https://github.com/PX4/ecl/pull/586

dagar, 2019-03-08:
We'd need to make sure they're properly marked as invalid.

CarlOlsson, 2019-06-24:
Still an issue

dagar, 2019-06-24:
You should be seeing `estimator_status` from the beginning (which has the states). Are you talking about also wanting to see `vehicle_attitude` and `vehicle_local_position`?

CarlOlsson, 2019-06-25:
yeah I was thinking about the output from the output predictor data before tilt align. But it is a very specific thing and I guess is not relevant enough to keep this open for now

closed 2019-06-25 by CarlOlsson