PX4/PX4-Autopilot #11620: Update RTL behavior description

AuterionWrikeBot, 2019-03-11:
@tubeme Is that still relevant?

[image]
 What will be the expected behaviour of the aircraft?

Does this mean that the aircraft will engage in Loiter at 30 meters then after 10 seconds will go out of the loiter transition and then descend to land? OR it will loiter for 10seconds and then land in a FW mode style going to the ground? Cannot find any info on this. 

We cannot use the RTL_TYPE 1 and 2 options because our landing sites are at different location than our home position.

hamishwillee, 2019-03-11:
FWIW I'd guess that it is, though there isn't enough information about the problem to be sure.

What the current/above dialog reflects is 
- [RTL_TYPE](https://docs.px4.io/en/advanced_config/parameter_reference.html#RTL_TYPE)==1 (fly home on direct path and then land or loiter based on other params). 
- [RTL_LAND_DELAY](https://docs.px4.io/en/advanced_config/parameter_reference.html#RTL_TYPE) - whether to land or just loiter - at `RTL_DESCEND_ALT`
- [RTL_DESCEND_ALT](https://docs.px4.io/en/advanced_config/parameter_reference.html#RTL_DESCEND_ALT) - altitude to drop to before deciding whether to land or loiter.

So what it means is that the vehicle will as part of landing descend to `RTL_DESCEND_ALT`, loiter for 10 seconds, then land. AFAIK The land behaviour depends on the vehicle. For a VTOL my understanding is that with this setup it will always transition to MC for the actual landing.

I think it is possible that the question is how can we use this landing ap

So what needs to change? I don't know, because this is posted as a firmware issue rather than a QGC issue and the problem to solve isn't clearly stated. I THINK that this requires a fix to QGC to reflect the new behaviour of [RTL_TYPE](https://docs.px4.io/en/advanced_config/parameter_reference.html#RTL_TYPE). This now has options 1, 2 which instead of just following the loiter/land behaviour above instead use land behaviour as defined in the current mission (they differ in the path back).

> We cannot use the RTL_TYPE 1 and 2 options because our landing sites are at different location than our home position.

@dagar Does this issue make sense to you? Could be that they want rally points, or that the mission landing would allow you to specify an alternative to home as the landing point?

tubeme, 2019-03-11:
Thanks @hamishwillee That is still relevant and that is my question. But are we sure after the loitering time passes it will transition to MC and land.

Before it was implemented so that the aircraft will jsut loiter and then land as FW aircraft. I dont know if this changes or not.

hamishwillee, 2019-03-11:
Hi @tubeme 

Actually my point was that I don't understand the question - unless it is these two
1. "please explain landing behaviour options on VTOL"
2. Can I specify an alternative land location to home for VTOL using mission landing

> Before it was implemented so that the aircraft will just loiter and then land as FW aircraft. I dont know if this changes or not.

That surprises me. According to https://docs.px4.io/en/flight_modes/land.html#vtol the vehicle should transition to MC by default. In theory the transition should depend on NAV_FORCE_VT but I distinctly remember @dagar telling me that NAV_FORCE_VT was not actually implemented so VTOL would always land as MC. That was some time ago. 

Daniel - can we have a chat about what mission-based landing, and in particular for VTOL. https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_VTOL_LAND it

hamishwillee, 2019-03-20:
@bkueng As above, not entirely clear to me the problem (and hence solution here). I think it is a request for VTOL to support alternate landing location for home in a mission.

But what is clear is that the QGC dialog does not reflect [RTL_TYPE ](https://docs.px4.io/en/advanced_config/parameter_reference.html#RTL_TYPE) - ie the dialog does not support using a mission landing or a return path other than direct. 
I can't find RTL_TYPE on search in qgc repo. Shouldn't there be an issue for this in QGC?

hamishwillee, 2019-06-27:
@bkueng If any thoughts on the above?

bkueng, 2019-06-27:
Best to let @RomanBapst answer, I don't know the specifics.

julianoes, 2020-01-30:
Closing because no answer.

closed 2020-01-30 by julianoes