PX4/PX4-Autopilot #11624: Loaded firmware for board id: 50,0 size: 1657824 bytes (80.31%), waiting for the bootloader

NiranjanRavi1993, 2019-03-12:
Hi,
    I have a **pixhawk 4**  board with me. I am using **px4 flight stack**. I was flashing my code into the device. It was working fine.
But all of a sudden this started happening:

**niranjan@niranjan-Predator-G3-571:~/Firmware$ make px4_fmu-v5_default upload
ninja: Entering directory `/home/niranjan/Firmware/build/px4_fmu-v5_default'
[2/3] uploading px4**
**Loaded firmware for board id: 50,0 size: 1657824 bytes (80.31%), waiting for the bootloader...**

It keeps on waiting and no response from the board. I tried changing the USB, unplugging and re-plugging. It did not help. Still, the issue persists.

I then opened **Qgroundcontrol**. This time the board got detected the upgrade was successful. But my code changes were not present. The simple_app fie was also not shown up in the **Mavlink console.** 
I am pretty sure my code is not getting dumped to the board.
 How do I fix it? I saw earlier issues regarding this and I tried the above troubleshooting methods.
But nothing helped.
Can someone please help me solve this issue?
Regards
Niranjan

closed 2019-06-24 by NiranjanRavi1993

Bunborn, 2019-10-24:
Hi, did you ever find a solution?

NiranjanRavi1993, 2019-10-25:
Yes, it was a quick fix. Take a fresh patch and redo the process but upload the binary file through ground control software.

ljemakers, 2020-07-29:
Hi, I'm going through the same thing as you. I'd like to get some help on how you solved it.

[image]

jmurraylouw, 2021-08-06:
Did anybody find a different solution?
I have the same problem with the Pixhawk4 Mini.

It does not work to use v3 or v2.

vibes99, 2021-09-03:
Hi, I'm also going through a similar problem. Has anyone been able to resolve it? 
Currently using Pixhawk4 mini board and trying to flash the code onto the controller.

Juno-cyber, 2022-09-06:
meet this problem too,I can upload .px4 file from QGC,but can not easily use make.. command.
it's not convenient,