PX4/PX4-Autopilot #11625: sending global position in offboard mode

garfieldG, 2019-03-12:
Hi,

Posting this issue again, the first one closed automatically with no response https://github.com/PX4/Firmware/issues/10917

I encountered a situation in which my drone reports it's position in global ( using POSITION_TARGET_GLOBAL_INT message), but receives position commands from offboard in local (SET_POSITION_TARGET_LOCAL_NED).
Because of the situation POSITION_TARGET_GLOBAL_INT that is reported to the GCS contains zeros in lat_int, lon_int and alt fields.

I investigated it a little and noticed that it happens due to the fact that while the fcc is in offboard mode the only place that position_setpoint_triplet is published from is mavlink_receiver when receiving SET_POSITION_TARGET_LOCAL_NED message and there is no reference to current.lat, current.lon and current.alt there.

julianoes, 2019-07-09:
Thanks for raising this. So you are saying that lat/lon here are not populated here?

https://github.com/PX4/Firmware/blob/59c555aec36aa5b1c1edac8d0d09fb90a36a3722/src/modules/mavlink/mavlink_messages.cpp#L3141-L3150

garfieldG, 2019-07-16:
@julianoes Exactly, only with zeros.

PR #12819 (2019-09-03): Enable support for global position setpoints with SET_POSITION_TARGET_GLOBAL_INT
**Describe problem solved by the proposed pull request**
Currently offboard position setpoints can only be passed in local coordinates with `SET_POSITION_TARGET_LOCAL_NED`. However, when running multiple vehicles at the same time, this is not scalable as setpoints need to be transformed to each vehicle's local frame.
Also, there has been a lot of workarounds for external applications to use global coordinates and converting it to local coordinates to just enable offboard mode.

Fixes #11625

**Describe your prefer [... cut]
closed 2019-09-03 by julianoes