PX4/PX4-Autopilot #11635: Lidarlite v3 driver does not automatically start with I2C connection

doraepon, 2019-03-13:
Hi,

Somehow when I connect lidarlite v3 using I2C, the lidarlite driver does not automatically start. I have to manually start it using nuttshell in QGC->mavlink console with the following command :
nsh> ll40ls start i2c 

I follow the wiring using the link below :
https://docs.px4.io/en/sensor/lidar_lite.html

For the param, I set param SENS_EN_LL40LS to 2 (I2C).

My pixhawk firmware is version "v1.8.2 Stable Release"

Right now I switch back to pwm and there is no problem. 

Thanks!

mcsauder, 2019-07-09:
Hi @doraepon , in v1.8.2, (on fmu-v5 for me) I get an error at boot:
```
ERROR [ll40ls] failed to initialize LidarLiteI2C on busnum=1
ERROR [ll40ls] failed to initialize LidarLiteI2C on busnum=2
ERROR [ll40ls] failed to initialize LidarLiteI2C on busnum=4
ERROR [ll40ls] failed to initialize LidarLiteI2C on busnum=3
WARN  [ll40ls] No LidarLite found
```

Upgrading to v1.9.2 solves this issue for me and the driver starts properly at boot.
```
LL40LS on I2C bus 2 at 0x62 (bus: 100 KHz, max: 100 KHz)
```

Can you upgrade to v1.9.2 and let us know if this resolves your issue?

mcsauder, 2019-07-30:
I think this issue is safe to close as it is fixed in v1.9.2.  @doraepon , please reopen if the issue persists for you with v1.9.2!

closed 2019-07-30 by mcsauder